Steffen-W/Import-LIB-KiCad-Plugin #2: MacOS not working

nemccarthy, 2023-05-20:
I noticed in the README that MacOS support is still tbd. Let me know if you would like a hand testing as this plugin looks awesome. Sadly do all dev on a mac and when clicking the plugin button nothing happens. Happy to help test/debug.

Steffen-W, 2023-05-28:
Hi @nemccarthy 
I'm trying this in macOS right now too. Unfortunately I have to agree with you that the desired window does not open. However, something opens in the background, which prevents KiCad from closing completely, at least for me. I am unfortunately clueless about macOS, but trying to locate the problem.

Steffen-W, 2023-06-03:
Hi, the problem seems to be that I have set the GUI to wxFrame and not wxDialog. Under Windows and Linux this is unproblematic. Why it doesn't work on Mac I don't know. But a change to wxDialog seems to make sense. Then it should work for you too.

nemccarthy, 2023-06-10:
Thanks @Steffen-W this is great news, how do I make the change for `wxDialog`?

Steffen-W, 2023-06-20:
Hi @nemccarthy,

I have customized the code so that it should run on Mac. I tested it at least briefly under Linux and MacOS. I will create a release sometime in the near future, because the window does not stay open in the background. Otherwise, this would be a step backwards from the previous version. I would be happy if you could take a look at the version.

Steffen-W, 2023-06-21:
I will test the last version and release it for everyone, if I don't find any bugs. Hopefully it will work on all platforms.
[v0.3.0](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/v0.3.0)

Steffen-W, 2023-07-07:
The plugin should have been working on Mac for some time. Should there be any problems, this can be reopened.

closed 2023-07-07 by Steffen-W

steeley, 2024-05-02:
using Kicad 8.02 on OSX 13.6.6. plugin installed ok, global paths ok. I select my import and save location paths, but when I press start nothing happens. the impartGUI does nothing.

steeley, 2024-05-02:
I have a library folder from Samacsys
[image]

window like this:
[image]

Steffen-W, 2024-05-02:
I have just tried the plugin. I noticed that the downloaded file has been unpacked directly and is not a zip file. Could this also be the case for you? In that case, you can pack it as a zip file as a test.

Steffen-W, 2024-05-02:
I don't know why this is unpacked after the donloade but pack it as a zip file. It must be a newer behavior.

Steffen-W, 2024-05-02:
https://www.wonderplugin.com/wordpress-tutorials/how-to-stop-safari-from-automatically-unzip-downloaded-zip-files/

steeley, 2024-05-02:
ok so I can get something from the .zip. Your FAQ says libraries do not need to be created first, but the import complains that .pretty is missing, but you cannot create this  file. Sorry this is all very confusing and makes no sense to me... 

Import: /Users/ian/Downloads/alib/LIB_THAT4305Q16-U.zip
Identify Samacsys
Import of kicad_sym skipped
Import of 3d model skipped
Import of footprint skipped
OK

##############################

Important information: 
If you have already used the previous version of the plugin, you should note that the current version supports all library files. Files with the new format are imported as *_kicad_sym and must be included in the settings (only Symbol Lib). The settings are checked at the end of the import process.

‘${KICAD_3RD_PARTY}/Samacsys_kicad_sym.kicad_sym’ is not imported into the Symbol Libraries.
Samacsys is not in the Footprint Libraries.
You have to import the library ‘Samacsys’ with the path ‘${KICAD_3RD_PARTY}/Samacsys.pretty’ in the Footprint Libraries.

More information can be found in the README for the integration into KiCad.
github.com/Steffen-W/Import-LIB-KiCad-Plugin
Some configurations require a kicad restart to be detected correctly.

##############################

Steffen-W, 2024-05-02:
[image]

It looks to me as if the import has already worked for you. Have a look.

steeley, 2024-05-02:
thanks - I'm not going to use your plugin. it is nor working as I have explained. sorry!

Steffen-W, 2024-05-02:
But it looks to me like it works. You have probably not set KICAD_3RD_PARTY. I have tried the function on a fresh mac and it is as desired.  [YouTube - Instructions for installation and use](https://youtu.be/BYIKjCs1qKQ)