Steffen-W/Import-LIB-KiCad-Plugin #3: Import-Lib installed, but not appearing anywhere in KICAD

SimGas, 2023-07-06:
Hi,

I am new to KICAD, so it could also be that I am just blind...
I've just installed your plugin as it seems to be exactly what I want. However, after pressing "apply changes" and restarting KICAD I can't find any of the buttons or functions of the plugin anywhere in KICAD. Neither in the schematics editor nor in the pcb editor.
Is there something I need to activate additionally?
Where can the Import-LIB dialog be accessed normally? I do not find it anywhere in the menus. 
If this is not an bug, I am sorry for opening an issue report here. However, the "Documentation" should be updated with all necessary steps (including where to find the import dialog) ;)

Thanks a lot,
Simon

PS: I am using KICAD 7.0.5 (Build May 27, 2023)

Steffen-W, 2023-07-06:
Hi,

what operating system are you using?
You should find the button in the pcb editor on the top right of the bar. A reboot after installation is not even necessary. https://www.youtube.com/watch?v=cdOKDY-F4ZU in second 8 you will see the button.

SimGas, 2023-07-07:
Hi Steffen,

thanks for your answer. I am using Windows 10, 22H2,
Yes, I watched the video in 0.25 Speed (in original Speed its really hard to see..) but I do not have that button. Please find a screenshot here:
[image]

Thanks
Simon

Steffen-W, 2023-07-07:
Hi,

i luckily found a PC with win 10 22H2. On this it runs with KiCad 7.0.5. Can you try to install another plugin. For example InteractiveHtmlBom (if you don't know it, it's worth to use it). 
If this has the same problem and does not show up, that would be a problem.
I have changed the GUI to v0.3.0, you could try the previous version if it works.
Otherwise it is also possible that only the button does not appear but the plugin still works. In PCB Editor Tools -> External Plugins -> impartGUI. :)

Please write me if I could help you with this. I would have to be able to reproduce the error myself to find it. Currently I can not do it.

Greetings
Steffen-W

SimGas, 2023-07-07:
Hi Steffen. 

It turned out that it did not even appear in the menu, external plugins you have just mentioned. HTMLGUI worked fine. So I tried reinstalling your plugin and now it is there. I have no idea why the installation failed before, but I also noticed that it only showed a generic Icon instead of the correct one. 

Problem solved. Sorry for bothering you.

closed 2023-07-07 by SimGas

Steffen-W, 2023-07-07:
No problem. I am happy that the problem has been solved. I'm sure there are other users who had the problem and did not write it. Hopefully the plugin can take work off your hands and not make you another one. The video what you watched in 1/4 speed, I will add a more detailed one soon. I am aware that the video is not that helpful.