Steffen-W/Import-LIB-KiCad-Plugin #5: Import of 3d models needs to use different 3rd party path variable for kicad 7

PlasmaHH, 2023-08-24:
Hi,

upon importing parts I noticed that 3d models where not availalbe in kicad7. The Path used was using ${KICAD_3RD_PARTY} while kicad7 uses KICAD7_3RD_PARTY.

Guess KiCadImport.py:312 needs to check for the kicad version that the plugin is running under....

Steffen-W, 2023-08-24:
Hi I unfortunately do not understand exactly what the problem should be. The variable KICAD_3RD_PARTY is so named by me and actually independent of the KiCad version. Which problem has arisen with you exactly?

PlasmaHH, 2023-08-24:
Oh, ok, my bad then, I thought that it would refer to the kicad built in KICAD7_3RD_PARTY path like in other plugins, guess I will just change it locally for me then.

closed 2023-08-24 by PlasmaHH