Steffen-W/Import-LIB-KiCad-Plugin #6: 3D Models getting two possible entrys

Stimmenhotel, 2023-09-20:
Hey Steffen, thanks for this handy plugin!
I amjust getting started with KiCad and really like you plugin over the library loader of Samacsys/Mouser.

I tried different ways yesterday to install and configurate the impart gui for KiCad7.07.
Most things were working fine after understanding the NAME of the lib is needed to link the footprint to the symbol and the 3rdparty PATHs are two different ones.

But what I haven't got working is a problem free 3d model import/link/view/whatever.

Symbol and footprint are properly linked, but once I open 3D-View or the Properties->3D-View, the part has two entrys.
1. part.extension 
2. {KICAD_3RD_PARTY}/Samacsys.3dshapes/part.extentsion 
[image]

The not working one (red X) has the values already included.
The shown model is the second one, missing the values and resulting missplaced.

Is there some configuration I missed?
I could change the KICAD7_3DMODEL_DIR to the one I added with impart, but now it shows two models:
[image]

Right now I help me by just copying the second path into the first one and deleting the second path afterwards.

Any tip for something more automatic would be appreciated.

Steffen-W, 2023-09-20:
Hi, I am glad that the plugin helps you. Unfortunately, I already know the error and have not yet fixed it. I am still surprised that you get both 3D models displayed. Nomally it does not find the upper entry and stallte only the lower there. You can at least solve the problem manually. It is not complex and hopefully unnecessary in the future.

The section to be modified: [Import-LIB-KiCad-Plugin/plugins/KiCadImport.py](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/5e2632ff3f1c6dd5d81a2da617605d3ed84ada7c/plugins/KiCadImport.py#L310-L349)

Go to the following file: ${KICAD_3RD_PARTY}/Samacsys.pretty/7865547.kicad_mod

```
...
  (pad 68 thru_hole circle (at -41.910 1.9) (size 1.118 1.118) (drill 0.61) (layers *.Cu *.Mask))
  (pad MH1 thru_hole circle (at 8.000 0.952) (size 4.155 4.155) (drill 2.77) (layers *.Cu *.Mask))
  (pad MH2 thru_hole circle (at -49.910 0.952) (size 4.155 4.155) (drill 2.77) (layers *.Cu *.Mask))
  (model 786554-7.stp
    (at (xyz -0.82480317964329 -0.037401574333822 0.062204726099029))
    (scale (xyz 1 1 1))
    (rotate (xyz -90 0 0))
  )
  (model "${KICAD_3RD_PARTY}/Samacsys.3dshapes/786554-7.stp"
    (offset (xyz 0 0 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 0))
  )
)
```

change it to:
```
...
  (pad 68 thru_hole circle (at -41.910 1.9) (size 1.118 1.118) (drill 0.61) (layers *.Cu *.Mask))
  (pad MH1 thru_hole circle (at 8.000 0.952) (size 4.155 4.155) (drill 2.77) (layers *.Cu *.Mask))
  (pad MH2 thru_hole circle (at -49.910 0.952) (size 4.155 4.155) (drill 2.77) (layers *.Cu *.Mask))
  (model "${KICAD_3RD_PARTY}/Samacsys.3dshapes/786554-7.stp"
    (at (xyz -0.82480317964329 -0.037401574333822 0.062204726099029))
    (scale (xyz 1 1 1))
    (rotate (xyz -90 0 0))
  )
)
```

If you make the changes in the file, they are saved permanently. You delete the entry you don't want and apply the settings for the shifts and rotations.

I hope I could help you with this. Good success with your KiCad project.

Stimmenhotel, 2023-09-26:
Thanks for your reply.

The second model is shown only, if I copy the path of the second entry (installed/created by your tool) into the first entry (in the picrure not done).
This way one model (the new copy) has the right settings, while the second still misses them.

Instead of editing the files directly, I just use the properties window to copy the path and delete the new entry. As far as I see, I just need to do this once too.

Steffen-W, 2023-09-26:
Hi @Stimmenhotel , you can of course do it directly in the properties window. But the problem is, if you reload the library and overwrite your local setting, then you may do it again. Therefore I would rather edit the entry once.
But I can do it soon and fix it in the code. I will close the issue as soon as I have programmed a solution.

Steffen-W, 2023-10-14:
Hi @Stimmenhotel,

The problem should be fixed. I would be happy if you test it with yourself.
[2023.10.14](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2023.10.14)

closed 2023-10-14 by Steffen-W

Stimmenhotel, 2023-10-20:
Thanks for the update, I just installed the latest version 2023.10.20

The parts I had problems with seem to work fine now.

The Wago 250-716 connector taken from snapeda seems to be shitty from the beginning. The symbol looks wrong, 3D seems to be mirrored and wrongly placed. But I really think, this is a problem at snapeda side.
For reference https://www.snapeda.com/parts/250-716/WAGO/view-part/?ref=search&t=wago%20250-716#

Is it needed to create the kicad_sym librarys beforehand? The text in the gui is not very clear about that.
Well I created the path for the libs and it works perfectly now.

Steffen-W, 2023-10-20:
Hi @Stimmenhotel ,

Unfortunately, the components from snapeda are not so nice. The alignment of the step file is unfortunately also not always the same, it feels random. I can not change this with the plugin. I also really have no idea how I could do this automatically.
Your hint with the new format is good. As with the other files, it is not necessary to create them. Under certain circumstances, it can even lead to errors if the file is not created correctly. Let the plugin create it for you and just import the created files.

For further hints of open questions I am very grateful.