Steffen-W/Import-LIB-KiCad-Plugin #7: Problems installing your plug-in

CR2O7, 2023-10-19:
Please see the attachment.
[Problems installing KiCad Plugin.odt](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/files/13042972/Problems.installing.KiCad.Plugin.odt)

Steffen-W, 2023-10-19:
Hello,

I would ask you to put your request in text and not in an odt document.

I will summarize briefly. The plugin appears to have been fully installed and the paths included as in the readme.

[image]

Steffen-W, 2023-10-19:
Hi @CR2O7 ,

In principle, you have done everything correctly. It can't find the libraries yet, because you haven't imported any library yet. You only need to import the ones that will appear in your directory.
The plugin should work normally for you. Have you tried it?

CR2O7, 2023-10-20:
Hi again,
Being a newcomer to KiCad, I don't understand your reply. If there are more steps I need to take apart from the once I've already done to get rolling, please inform me.
Specifically, what shall I do with this: "You only need to import the ones that will appear in your directory".

Rgds,
Göran

Steffen-W, 2023-10-20:
Hi @CR2O7 ,

I have adapted the readme and created a new version of the plugin. I would be happy if you want to test it. Just download and choose install from file in plugin manager. Is it now more understandable for you in the README? When you use the plugin, are the messages you get detailed enough and do you feel you understand what is happening?

[2023.10.20/Import-LIB-KiCad-Plugin.zip](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/download/2023.10.20/Import-LIB-KiCad-Plugin.zip)

CR2O7, 2023-10-20:
Ok, let's go through this in detail.

First, my expectations is after having properly installed the your stuff by
using one of the suggested methods, "Install from File... or simply press
the "Install"-button I should get all symbols, footprints etc into my
KiCad, or?

I've tries both of these methods and they seems both to be working, the
button in the "Plugin And Content Manager" for the "impart GUI for Kicad"
entry changes from "Install" to "Uninstall".

In your README.md under the heading "Use of the application" is there
something I suppose to run, but how?

Furthermore, the text: "
*(Note: It is best to add the library only after the import has been done
with the plugin. Afterwards only the created libraries have to be imported.
Lower entries are only for example.)" *

Or if they are already typed in I can leave  them as they are and later if
decided not to import from a source just delete these entries?

Regards,
Göran

Steffen-W, 2023-10-20:
Hi @CR2O7 ,

I'll try to answer a few questions. Definitely more than you have asked. But with one or the other they come up.

**Where can I save the library?**
It is possible to save the files in any place on your PC where you have read and write permissions.
It is also possible to store the files on network drives or cloud storage to share the library with others.
In the libraries relative paths are used, the absolute path is not considered.

**Can I change the storage location?**
Yes, this is of course always possible. But you should keep in mind that the existing libraries will not be moved automatically. You would have to do that yourself if necessary.

**If I import from one source, do I have to stay with that source, or can I import from all sources?**
For each source, a separate library is created for KiCad. Maximum actually three (Samacsys, Ultralibrarian and Snapeda), Octopart components as well as from other suppliers can be found in the Snapeda library. So if you import from a new source, a new library can be created. But maximum three.

**Why are there two symbol libraries?**
In the latest version both the old and the "new" (since KiCad 6) symbol library format is imported. It is possible to prevent this by deselecting "import also old format". An automatic conversion from the old to the new format should only be done if you are an experienced KiCad user.

**General KiCad question:**
**I have entered a library in the settings in KiCad that does not exist at this time, what happens?**
You get a message that this library does not exist. Nothing more. As soon as it exists, it can be accessed.

**Can I remove a library from the settings?**
Yes, you can always do that. The libraries are neither deleted nor edited in any way. They are just not included.

It's always good to get feedback because you like to get caught up in your own thoughts and experiences. The experience of a not-so-experienced user is unfortunately lost over time.

closed 2023-10-20 by Steffen-W

CR2O7, 2023-10-20:
I notice you answered questions I didn't ask and didn't answered question I did ask, so here is the key question:

From where or how do I run the plug-in that will do the imports?

Steffen-W, 2023-10-20:
I hope that I understand the question correctly now. You specify two locations in the plugin. From where it will be imported "folder of the library to import" (mostly your download folder) and where it will be saved "library save location". The place is free to choose.

CR2O7, 2023-10-21:
Now it is just some details left...

One critical piece of information is missing in the instructions, from
where to run the plugin:
[image: Screenshot from 2023-10-21 08-48-35.png]
When I run plugin the following message is shown:
"

*Import:
/home/linuxgoran/CloudStation/KiCad/KiCad_imports/LIB_ISO1050DUBR.zip*

*Identify Samacsys*

*Import kicad_sym*

*Import lib*

*Import 3d model*

*Import footprint*

*OK*

*##############################*

*Important information: *

*If you have already used the previous version of the plugin, you should
note that the current version supports all library files. Files with the
new format are imported as *_kicad_sym and must be included in the settings
(only Symbol Lib). The settings are checked at the end of the import
process.*

*'${KICAD_3RD_PARTY}/Samacsys_kicad_sym.kicad_sym' is not imported into the
Symbol Libraries.*

*Samacsys in the Footprint Libraries is not imported correctly.*

*You have to import the library 'Samacsys' with the path
'${KICAD_3RD_PARTY}/Samacsys.pretty' in Footprint Libraries.*

*More information can be found in the README for the integration into
KiCad.*

*github.com/Steffen-W/Import-LIB-KiCad-Plugin
<http://github.com/Steffen-W/Import-LIB-KiCad-Plugin>*

*Some configurations require a kicad restart to be detected correctly."*

How do I rectify it?

Regards,

Göran

Steffen-W, 2023-10-22:
Hi @CR2O7,

The import process of the plugin ran without errors for you. The section with the information is only an info but not an error message. You have to import the library file as explained in the readme. It is written:

${KICAD_3RD_PARTY}/Samacsys_kicad_sym.kicad_sym' is not imported into the
Symbol Libraries.*

So add it to the symbol libraries. The same with:

${KICAD_3RD_PARTY}/Samacsys.pretty' in Footprint Libraries.

This means the settings in kicad. Read the readme. If you can think of a more understandable message that is intuitive to understand I would be grateful for suggestions.

I will also make a video where I demonstrate the plugin but I've been putting it off for a while.
Übersetzt mit DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

Steffen-W, 2023-10-22:
Hi @CR2O7,

The import process of the plugin ran without errors for you. The section with the information is only an info but not an error message. You have to import the library file as explained in the readme. It is written:

${KICAD_3RD_PARTY}/Samacsys_kicad_sym.kicad_sym' is not imported into the
Symbol Libraries.*

So add it to the symbol libraries. The same with:

${KICAD_3RD_PARTY}/Samacsys.pretty' in Footprint Libraries.

This means the settings in kicad. Read the readme. If you can think of a more understandable message that is intuitive to understand I would be grateful for suggestions.

I will also make a video where I demonstrate the plugin but I've been putting it off for a while.

If you have a new question, open a new issue. It will be easier for others to find if they have the same problem.