Steffen-W/Import-LIB-KiCad-Plugin #9: 'ascii' codec can't decode byte in Some Snapeda v6 files

xMAC94x, 2024-01-17:
Hi, importing of Some of my Snapeda v6 files failes with an ascii codec error.
The relevant files are:
 - https://www.snapeda.com/parts/150060RS75000/Wurth+Elektronik/view-part/
 - https://www.snapeda.com/parts/AZ1117CH-3.3TRG1/Diodes+Inc./view-part/
 - https://www.snapeda.com/parts/C2012X7R1E105M085AB/TDK/view-part/
 - https://www.snapeda.com/parts/GRM188R61E475KE11D/Murata+Electronics+North+America/view-part/
 - https://www.snapeda.com/parts/SFR10EZPJ101/Rohm/view-part/

here the detailed import protocol:
```
Import: /home/user/Downloads/kicad/06033C104KAT2A.zip
Identify Snapeda
Import of kicad_sym skipped
Import of 3d model skipped
Import of footprint skipped
OK

Import: /home/user/Downloads/kicad/150060RS75000.zip
Identify Snapeda
'ascii' codec can't decode byte 0xc3 in position 882: ordinal not in range(128)

Import: /home/user/Downloads/kicad/AZ1117CH-3.3TRG1.zip
Identify Snapeda
'ascii' codec can't decode byte 0xc2 in position 1737: ordinal not in range(128)

Import: /home/user/Downloads/kicad/BFP650FH6327XTSA1.zip
Identify Snapeda
Import of kicad_sym skipped
Import of 3d model skipped
Import of footprint skipped
OK

Import: /home/user/Downloads/kicad/C2012X7R1E105M085AB.zip
Identify Snapeda
'ascii' codec can't decode byte 0xc2 in position 756: ordinal not in range(128)

Import: /home/user/Downloads/kicad/CRCW080520K0FKEA.zip
Identify Snapeda
Import of kicad_sym skipped
Import of 3d model skipped
Import of footprint skipped
OK

Import: /home/user/Downloads/kicad/CRCW08054K70FKEA.zip
Identify Snapeda
Import of kicad_sym skipped
Import of 3d model skipped
Import of footprint skipped
OK

Import: /home/user/Downloads/kicad/GRM188R61E475KE11D.zip
Identify Snapeda
'ascii' codec can't decode byte 0xc2 in position 2254: ordinal not in range(128)

Import: /home/user/Downloads/kicad/ICM-42688-P.zip
Identify Snapeda
Import of kicad_sym skipped
Import of 3d model skipped
Import of footprint skipped
OK

Import: /home/user/Downloads/kicad/SFR10EZPJ101.zip
Identify Snapeda
'ascii' codec can't decode byte 0xc2 in position 722: ordinal not in range(128)

Import: /home/user/Downloads/kicad/TL3305BF260QG.zip
Identify Snapeda
Import of kicad_sym skipped
Import of 3d model skipped
Import of footprint skipped
OK

##############################

Important information: 
If you have already used the previous version of the plugin, you should note that the current version supports all library files. Files with the new format are imported as *_kicad_sym and must be included in the settings (only Symbol Lib). The settings are checked at the end of the import process.

KICAD_3RD_PARTY is not defined in Environment Variables.

More information can be found in the README for the integration into KiCad.
github.com/Steffen-W/Import-LIB-KiCad-Plugin
Some configurations require a kicad restart to be detected correctly.

##############################

```

Steffen-W, 2024-01-17:
Hi @xMAC94x ,

I have tried to recreate it and unfortunately cannot get the error.
```
Import: /home/user/Downloads/150060RS75000.zip
Identify Snapeda
Import kicad_sym
Import 3d model
Import footprint
Warning renaming footprint file "LED_150060RS75000" to "150060RS75000_LED_150060RS75000"
OK

Import: /home/user/Downloads/AZ1117CH-3.3TRG1.zip
Identify Snapeda
Import kicad_sym
Import 3d model
Import footprint
Warning renaming footprint file "DIODES_AZ1117CH-3.3TRG1_0" to "AZ1117CH-3.3TRG1_DIODES_AZ1117CH-3.3TRG1_0"
OK

Import: /home/user/Downloads/C2012X7R1E105M085AB.zip
Identify Snapeda
Import kicad_sym
Import 3d model
Import footprint
Warning renaming footprint file "CAPC2012X100N" to "C2012X7R1E105M085AB_CAPC2012X100N"
OK

Import: /home/user/Downloads/GRM188R61E475KE11D.zip
Identify Snapeda
Import kicad_sym
Import 3d model
Import footprint
Warning renaming footprint file "CAPC1608X95N" to "GRM188R61E475KE11D_CAPC1608X95N"
OK

Import: /home/user/Downloads/SFR10EZPJ101.zip
Identify Snapeda
Import kicad_sym
Import 3d model
Import footprint
Warning renaming footprint file "RESC2012X65N" to "SFR10EZPJ101_RESC2012X65N"
OK
```

Which option did you select when downloading? I have selected "KiCad / V6 & Later" at Snapeda. Which Kicad version as well as which plugin version. What system is KiCad running on and in which language?

Regards Steffen-W

Steffen-W, 2024-02-20:
I assume that the error has been resolved.

42CrMo4, 2024-02-21:
@Steffen-W I'm facing the same errors as @xMAC94x. On my system I also can reproduce the messages from the original post with the mentioned Parts. 

For example the:
https://www.snapeda.com/parts/AZ1117CH-3.3TRG1/Diodes+Inc./view-part/
Downloaded as Kicad v6 & later. 

Error Message:
```
Import: /Users/tobias/Downloads/AZ1117CH-3.3TRG1.zip
Identify Snapeda
‚ascii‘ codec can’t decode byte 0xc2 in position 1737: ordinal not in range(128)
```

looking through the *.kicad_sym file there is the following character U+00b0 "°" that is the problem. One example: 
```
(property "TEMPERATURE_RANGE_LOW" "-20°C" (id 12) (at 0 0 0)
      (effects (font (size 1.27 1.27)) (justify bottom) hide)
    )
```
visual studio code warning message: 
```
The character U+00b0 "°" is not a simple ASCII character.
```

after deleting every instance of this character and creating a zip again the import works fine. 

My system comprises of the following versions: 
```
Application: KiCad PCB Editor
Version: (6.0.10-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/8.1.2 SecureTransport (LibreSSL/3.3.6) zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Version 13.6.1 (Build 22G313), 64 bit, Little endian, wxMac

Build Info:
	Date: Dec 18 2022 11:46:04
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.3
	Curl: 7.64.1
	ngspice: 38
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

impart gui for kicad:
        2023.10.20
```

Please let me know if i should open a new issue instead of appening to this one. 
Also if you need any further information testing I'm happy to help.

Steffen-W, 2024-02-21:
Hi @42CrMo4,

thank you very much for your detailed description of the problem. I have now taken a closer look at it.

‚ascii‘ codec can’t decode byte 0xc2 in position 1737: ordinal not in range(128)

name: DEGREE SIGN "°"
Unicode: U+00B0
UTF-8: c2 b0

My Setting: de_DE.UTF-8 and Linux Mint

What is output when you do the following?
```python
import sys
print(sys.getdefaultencoding())
```
is it ascii?

KiCad seems to have no problem with the characters. It is only Python. Unfortunately I don't have a solution at the moment.

42CrMo4, 2024-02-21:
Thank you for taking another look. 

if i run in the kicad terminal 
```
import sys
print(sys.getdefaultencoding())
```
i get utf-8

No, prolem. I will also take a look on another PC.

42CrMo4, 2024-02-21:
Just a quick check, i did the same with KiCAD 7.0.10 . unfortunately the same result.

42CrMo4, 2024-02-21:
I think I found a solution that fixed it, at least for me. Specifing the encoding in the read_text() functions. 
```
read_text() 
to 
read_text(encoding='utf-8' )
```
If you want, I can provide a pull request tomorrow.

PR #12 (2024-02-24): Specify encoding as utf-8
closes #9 

Open a file or read text in utf-8 encoding.
closed 2024-02-24 by Steffen-W