Steffen-W/Import-LIB-KiCad-Plugin #11: New library format files named weirdly

abauske, 2024-02-19:
Hi Steffen,

thanks for the great plugin and all your work!

I was wondering whether it is intentional, that KiCad new library format is imported into library files that are called `<Source>_kicad_sym.kicad_sym` , wouldn't `<Source>.kicad_sym` be enough?
Moreover when importing into a blank folder (clear all previously added libraries in that folder) then I get `Samacsys.lib/dcm` files but also `Samacsys_kicad_sym.dcm` and `Samacsys_kicad_sym.dcm~` (mind the tilde) and `Samacsys_kicad_sym.kicad_sym`. Also for Snapeda I am getting `Snapeda_kicad_sym.dcm` and `Snapeda_kicad_sym.kicad_sym`, shouldn't `Snapeda.dcm/kicad_sym` be fine?

Is this intentional? Am I missing something?

Steffen-W, 2024-02-20:
Hi @abauske ,

I understand your concern. You haven't done anything wrong. The naming of the files is due to the compatibility I wanted to ensure. In fact, it is a bug that the file "*_kicad_sym.dcm~" is not deleted in your case. But this is due to the code. I will assign the names properly when I have managed to convert the old library formats. Unfortunately this is currently not possible due to a KiCad bug. I wanted to have solved the problem a long time ago, but I haven't taken the time yet.

To summarize: Everything is fine with you and I fully understand your concern.

Steffen-W, 2024-08-03:
@abauske,

i am only writing this as an update and as a reminder to myself. The bug in kicad cli has now been fixed. I have already started to integrate it into the plugin.
[plugins/kicad_cli.py](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/kicad_cli.py)
I'm at the point where I don't know exactly what the best solution is right now. Some people have both the new and an old version of the libraries. Some only have the new version. I can now bring both versions up to date. Unfortunately, the problem is that I might have the components in two versions and would have to merge the libraries. What do you think? Should I ignore the legacy components and just move Snapeda_kicad_sym.kicad_sym to Snapeda.kicad_sym for the best game. Alternatively, should I keep the name. What do you suggest?

abauske, 2024-08-04:
To be honest the best way would probably to merge old and new and rename this to Snapeda.kicad_sym. However I think there is quite a bit of risk (in merging) and confusion / setup problems for the user.
Therefore I would probably leave old libraries as is and only for new imports move them to the new one. As updating the plugin is a manual process we could maybe allert the user here to rename/merge or at least add least add the new library location to the list of libraries? (which would need to be done for all other options likewhise)
What do you think? Is there a anything I have overlooked?

Steffen-W, 2024-08-08:
Now the strange name is no longer necessary. I only have to merge the two old formats.

closed 2024-08-08 by Steffen-W

Steffen-W, 2024-08-09:
Hi @abauske,
Hi, I have created a preliminary version for the conversion of the libraries. Currently no libraries are converted or edited. This is for security reasons, so there is an output in the command line what should be converted. If you have time I would be happy to hear your opinion on this.
[impart_action.py#L289-L321](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/impart_action.py#L289-L321)
I have created a prerelease for easy testing.
[releases/tag/2024.08.09](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2024.08.09)