Steffen-W/Import-LIB-KiCad-Plugin #13: Kicad Library Plug In

mystic12, 2024-02-25:
I found your github posted plugin for Kicad Library importing -- looks like just what I need.  I followed the instructions for installation and I had no problems. The plug was successfully "applied" in the Kicad plugin manager.  However, I can't find anyway to bring your plugin up for use.  The screen and example you show for using the program are no where to be found in Kicad -- I've looked all over.  Sorry for the "dumb" question, but how the hell does Kicad allow a user to bring up your library plugin?  Or any plugin for that matter?  Please write a simple, step by step instruction for Kicad 7.  I have used Kicad for years, but never needed a plugin.  The parts library problem has just gotten to big to ignore.
Also, if you happen to know of a good reference source or example of how to organize the various folders for Kicad 7 libraries, I would be VERY grateful.  I have used several PCB programs over 30 years, but Kicad is unique in its disorganized and complex file systems.

Thank you!
Patrick

Steffen-W, 2024-02-26:
Hi @mystic12 ,

[image]
If you have successfully installed the plugin, you should be able to find it in two places. I have marked it in the picture.

mystic12, 2024-02-26:
Steffen,
Okay, I can locate the plugin and it appears to be operable.  The problem is that the example runs so fast I can't figure out exactly what you are doing.  Example: I just downloaded from Mouser the CAD info for the Analog Devices ADSP-21565.  I get a zipped folder in my download box.  Does your plugin open up and extract directly from the zip file, or do I need to unpack?  When unpacked, there are files for all of the major PCB software, including Kicad.  Do I need to unpack and create a Kicad folder  to put the various files into?  Does it extract the footprint into (###.pretty) and symbol into (.sym) and the library into (.lib)?   Sorry, but your example runs too fast and assumes that the user (me) knows how to do this. 

Regards, Patrick

Steffen-W, 2024-03-02:
Hi  @mystic12,
I have created a video. I hope that it will help you.
https://www.youtube.com/watch?v=BYIKjCs1qKQ

closed 2024-03-02 by Steffen-W