Steffen-W/Import-LIB-KiCad-Plugin #14: 3D Models in 3rd party directories are not recognized

geeeethub, 2024-03-08:
This seems more of a KiCAD workflow issue than the plugin, but it most likely can be solved by allowing to specify where 3dmodels are stored.

When importing symbols / footprint / 3dmodel, the symbol and footprint go to their intended directories and the libraries and footprints show up as expected, but the 3dmodels are put into directories, such as:

```
${KICAD8_3RD_PARTY}/Samacsys.3dshapes/
${KICAD8_3RD_PARTY}/Snapeda.3dshapes/
```

Even though I added these as paths (i.e., KICAD_USER_3DMODEL_DIR = ${KICAD8_3RD_PARTY}/Samacsys.3dshapes/), KiCAD does not search those directories for the 3d models. 

If I specifically edit the path of the 3d file to be:

`${KICAD8_3RD_PARTY}/Samacsys.3dshapes/some3dmodel.stp`
or
`${KICAD_USER_3DMODEL_DIR}/some3dmodel.stp`

It works fine, but it is a pain to go into every footprint to do this when this plugin does so much already. I combed several forums and nothing seems to answer this question / issue directly.

Steffen-W, 2024-03-08:
Hi @geeeethub, 
it is a bug if the path is not entered. It should be "${KICAD_3RD_PARTY}/Samacsys.3dshapes/some3dmodel.stp". Which component do you have that did not work?

Steffen-W, 2024-03-08:
You can watch the video [(https://youtu.be/BYIKjCs1qKQ)](https://youtu.be/BYIKjCs1qKQ) and see that the components are correctly linked to the 3D model.

closed 2024-04-08 by Steffen-W

tenninebt, 2024-05-29:
I know the issue is closed, but it might be worth supporting both KICAD8_3RD_PARTY and KICAD_3RD_PARTY variables for better compatibility with all the version, the latest always take precedence or just map the variable(s) using KiCad version (if it is accessible to the plugin)
And thanks a lot for the plugin! 
Any plans to add a search module? (Snapeda official plugin is not compatible with the latest versions of KiCad) I might contribute if it is a direction you are willing to take.