Steffen-W/Import-LIB-KiCad-Plugin #15: Problem importing SNAPEDA parts into KiCAD 8.0

thekarlmiller, 2024-03-11:
**_Problem importing SNAPEDA parts into KiCAD 8.0_**
Date: **_3/11/2024_**
impartGUI version installed: **_2024.02.24_** 

For the above, the following steps were completed after installing the plugin using the KiCAD 8.0 "Plugin and Content Manager", before attempting to use the plugin:
**1**. from "Preferences --> Configure Paths", I created the Environment Variable named "KICAD_3RD_PARTY" - note, KiCAD 8.0 already created a similar variable named "KICAD8_3RD_PARTY" which points to a different folder.

**2**. from the "Schematic Editor --> Preferences --> Manage Symbol Libraries..." menu option, I added an "Empty Row" to the Global Libraries table with the following values:
	* Active - checked
	* Visible - checked
	* Nickname - Snapeda
	* Library Path - ${KICAD_3RD_PARTY}/Snapeda.lib
	* Library Format - Legacy
	* Options - Snapeda
	* Description - Snapeda symbols imported from "impartGUI" plugin

**3**. from the "PCB Editor --> Preferences --> Manage Symbol Libraries..." menu option, I added an "Empty Row" to the Global Libraries table with the following values:
	* Active - checked
	* Nickname - Snapeda
	* Library Path - ${KICAD_3RD_PARTY}/Snapeda.pretty
	* Library Format - KiCad
	* Options - 
	* Description - Snapeda footprints folder imported using "impartGUI" plugin

**4**. create folder associated with the ${KICAD_3RD_PARTY} environment variable at: C:\Users\theka\Documents\KiCad with the following entries:
	* Snapeda.pretty empty folder
	* partDownloads with downloaded PC1108.zip downloaded from Snapeda - it contains the following files:
	* 	how-to-import.htm
	* 	PC1108.kicad_sym
	* 	PC1108.step
	* 	TERMINAL_PC1108.kicad_mod

**5**. restart KiCAD 8.0 

**6**. from the PCB Editor I clicked on the impartGUI button
	6a. checked ONLY the "automatic import" chechbox
	6b. in the "Folder of the library to import:" I put the following text:  C:\Users\theka\Documents\KiCad\partDownloads
	6c. in the "Library save location:" I put the following text: C:\Users\theka\Documents\KiCad
		this is the same folder as the ${KICAD_3RD_PARTY} environment variable.
	6d. clicked on the "Start" button on the impartGUI dialog box.

**7**. After that, an WARNING dialog box without any title popped up with the following text:
	'${KICAD_3RD_PARTY}/Snapeda_kicad_sym.kicad_sym' is not imported into the Symbol Libraries.

	More information can be found in the README for the integration into KiCad.
	github.com/Steffen-W/Import-LIB-KiCad-Plugin
	Some configurations require a kicad restart to be detected correctly.

... so I clicked on the OK button.

**8**. The final text in the impartGUI dialogue box was:
	Import: C:\Users\theka\Documents\KiCad\partDownloads\PC1108.zip
	Identify Snapeda
	Import kicad_sym
	Import 3d model
	Import footprint
	cannot access local variable 'dcm_file_write' where it is not associated with a value

	##############################

	Important information: 
	If you have already used the previous version of the plugin, you should note that the current version supports all library files. Files with the new format are imported as *_kicad_sym and must be included in the settings (only Symbol Lib). The settings are checked at the end of the import process.

	'${KICAD_3RD_PARTY}/Snapeda_kicad_sym.kicad_sym' is not imported into the Symbol Libraries.

	More information can be found in the README for the integration into KiCad.
	github.com/Steffen-W/Import-LIB-KiCad-Plugin
	Some configurations require a kicad restart to be detected correctly.

	##############################

**9**. I click on the Stop button at the top of the impartGUI dialog box, then closed it.

**10**. Checking the C:\Users\theka\Documents\KiCad\ folder, it now contains the following ADDITIONAL files:
	* Snapeda.3dshapes\PC1108.step
	* Snapeda.pretty\TERMINAL_PC1108.kicad_mod
	* Snapeda.pretty\TERMINAL_PC1108.kicad_mod~
	* Snapeda_kicad_sym.dcm
	* Snapeda_kicad_sym.kicad_sym

**11**. restarted KiCAD 8.0

**12**. In the schematic editor, I pressed the A key (place symbol), and an error dialog box appeared with the following text:
	Errors loading symbols:
	  Error loading symbol library Snapeda.
	  Library file 'C:\Users\theka\Documents\KiCad\Snapeda.lib' not found.

**13**. after pressing the OK button for the error, that error doesn't happen on subsequent loads of the "Choose Symbol..." dialog box.
    The Snapeda library does show in the filtered items list, but there is no entries in it, and the PC1108 symbol is not found.

Can you please let me know what I am doing incorrectly?
I can provide the files if necessary.
Thank you,
THE Karl Miller

Steffen-W, 2024-04-07:
Hi @thekarlmiller,

Sorry for the very late reply. Unfortunately I was away for a very long time. It is a bug. I still hope that you were able to use the component. The DCM file is actually optional and is not absolutely necessary.

Steffen-W, 2024-04-08:
Hi @thekarlmiller , 

The error should be fixed. 
It would be very nice if you would test the version [2024.04.08](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2024.04.08) before I distribute it via kicad.

closed 2024-04-08 by Steffen-W

mbdriscoll, 2024-04-21:
Hi @Steffen-W,

I can confirm version [2024.04.08](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2024.04.08) fixes the `dcm_file_write` error on KiCad 8.0.1. Thanks for the fix, and the plugin in general!