Steffen-W/Import-LIB-KiCad-Plugin #16: Footprint name in Symbol not correct

sinalco92, 2024-03-30:
If I implement parts from SnapEDA, the naming of the footprint in the symbol properties is not correct. The naming locks like: "Symbol_Footprint" in the symbol properties. It should be only the footprint name, without symbol name and underscore (see picture). 

[image]

Steffen-W, 2024-03-30:
Hi @sinalco92,

This is not an error, but a conscious decision on my part. I did it because I wanted the footprint and ultimately didn't care which chip was in the package. So if 10 chips have a sot package then the footprint is called sot and not with 10 different names.

sinalco92, 2024-03-30:
But than the footprint must be linkt every time by hand. 
Iam use KiCad 8.0.1 and MacOS. And I installed and configured the Plugin, exactly like in your Youtube Video

Steffen-W, 2024-03-30:
OK @sinalco92 , of course that shouldn't be the case. The correct footprint is not entered in the symbol. I'm currently on the road and will have a look at what could have gone wrong at the end of next week.

Steffen-W, 2024-04-08:
Hi @sinalco92,

The error should be fixed. 
Fixed with commit 560b05fe3c8e86a6276d6e99aae860a08f26bba0
It would be very nice if you would test the version [2024.04.08](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2024.04.08) before I distribute it via kicad.

closed 2024-04-08 by Steffen-W