Steffen-W/Import-LIB-KiCad-Plugin #17: Increase .zip size limit (certain files not detected)

AndBondStyle, 2024-05-11:
The [first symbol](https://www.snapeda.com/parts/ESP32-DEVKIT-V1/Do%20it/view-part/) I tried to import with your plugin was ~15 MB in size (zipped), so it took me a while to debug why it didn't detect it, and didn't even output any useful logs in the text box. Very confusing... So maybe you can increase .zip size limit, or at least add some extra logging?

https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/76f022a27d0dd714220fad4e4636bf4a073f3322/plugins/impart_helper_func.py#L30-L35

Steffen-W, 2024-05-12:
Hi  @AndBondStyle, I'm sorry that you've encountered this problem. It was somehow foreseeable that it would come up at some point. Can you still tell me which component it was? What would you suggest, what should I change? I can of course display a message that the following large files are being ignored, but it should only really help a few people. I have to find out somehow whether it is a component library to be imported or not. I had the problem that large zip files took a long time to search through.

Steffen-W, 2024-05-12:
Sorry, I didn't notice the link. “ESP32-DEVKIT-V1.step” is 16MB in size (and looks good). I can increase the limit to 25MB, but this will only postpone the problem. What do you think?

AndBondStyle, 2024-05-12:
@Steffen-W I think the proper way to fix this "forever", is to keep a database of seen (processed) .zip files and skip them

Steffen-W, 2024-05-12:
That's what I'm doing for the current session. But I can also do it permanently. If you think so, I can implement it. But I would still leave the limit with the size. But then I can increase it to 50MB or so.

Steffen-W, 2024-05-12:
Thank you for not giving up on the plugin after it didn't work right away.

AndBondStyle, 2024-05-12:
No problem, open source is not easy, I get it :)

Yeah, I was talking about a permanent database, meaning that even after kicad restarts, we don't process those files again. But to check it, filenames are not enough, the most reliable option is to compare hashes (md5 for example). Another option is to use filename + last modification time (but it's not very reliable on windows, as far as I know).

Steffen-W, 2024-05-12:
I program later that the processed file names from the previous session are used for initialization. This should then be a solution for most component libraries.

https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/76f022a27d0dd714220fad4e4636bf4a073f3322/plugins/impart_helper_func.py#L10

I can note the time of the last change, but I still have to bring it in. By the way, OpenSource is fun, especially when you receive such suggestions for improvement. I'll adjust the code today.

Steffen-W, 2024-07-11:
Unfortunately I don't have the time to write it performant yet so I leave the file limit at 50MB and close the issue. If someone feels called upon to take care of it, I would be happy to receive a merge request.

closed 2024-07-11 by Steffen-W