Steffen-W/Import-LIB-KiCad-Plugin #20: Action for release (quality of life update?)

42CrMo4, 2024-07-14:
Hi there, 

I don't know how much trouble this is, but I just wanted to point out that you could automate the release and metadata creation for the KiCad PCM with github action. Here are two repos I found that do this:

[espressif-kicad-libraries](https://github.com/espressif/kicad-libraries/blob/main/.github/workflows/release.yml)
[Bouni-kicad-jlcpcb-tools](https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/.github/workflows/kicad-pcm.yml)

If you are aware of this and do not need it, feel free to close this issue. 

Cheers.

Steffen-W, 2024-07-14:
Thank you very much,

I was not aware of this possibility. Admittedly, I have not yet sufficiently familiarized myself with the possibilities offered by github. But do I understand correctly that this is primarily about metadata.json in the repo on github?

42CrMo4, 2024-07-14:
Yes, I think the two repos I link to only use it to prepare the metadata.json and zip creation. But I think it should also be possible to interact with your gitlab account via an API access token. To automatically fork, edit and commit. If you're interested, I'll check. 

Github (or Gitlab) actions are super powerful and useful. For Kicad designs I can recommend [Kibot](https://github.com/INTI-CMNB/KiBot).

Steffen-W, 2024-07-21:
Sorry for the late reply. I had created [generate_zip.sh](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/generate_zip.sh) which made my life much easier. This also creates the metadata file, even if only in a zip file. As long as the repo file is cloned directly into kicad, I don't think it matters if the data in it is up to date. [generate_zip.sh](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/generate_zip.sh) already adjusts the corresponding lines.

I currently have other things that I would like to integrate into the plugin and would of course be happy if you would have fun implementing them with me. 

On the one hand, I would like to update the libraries to the current version or bring them up to date before I import them via “kicad-cli sym upgrade Samacsys.lib -o Samacsys.kicad_sym”. This would prevent me from having to import both the old and the new library pre-mat. It would have some advantages.

The other is that I would very much like to use the functionality of [easyeda2kicad.py](https://github.com/uPesy/easyeda2kicad.py) and simplify the import of easyeda files. 

Maybe you'd like to. Neither is overly complicated.

42CrMo4, 2024-07-22:
Hey, no problem. 
That is fair enough. So I will close this issue. I will see if I can address some of the topics mentioned, but I have to admit that my programming knowledge is very basic.

closed 2024-07-22 by 42CrMo4

Steffen-W, 2024-07-23:
[kicad_cli.py](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/kicad_cli.py)
The update process is already prepared and only needs to be integrated. You probably have to put the files to be imported into a temporary directory, then update them with kicad-cli and then import them into the existing structure. So copying would be the first step.

[easyeda2kicad.py](https://github.com/uPesy/easyeda2kicad.py)
I have packed the library into the plugin directory in the zip file ([generate_zip.sh](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/generate_zip.sh#L10-L15)). This should prevent any import problems. I have also already expanded the gui. Only the function `ButtomManualImport` needs to be written [impart_gui.py#L140](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/213a68f22e234927548cb7696d20632b7a879111/plugins/impart_gui.py#L140). 

If you feel like it feel free to collaborate I will check the code anyway and adjust it if necessary.