Steffen-W/Import-LIB-KiCad-Plugin #23: Error: No module named 'pydantic'

Alviunta, 2024-08-20:
I have an error using the import functions of "LCSC Part#".
When I try to run it with a part number, I get an error message "Error: No module named 'pydantic'".
I tried to install pydantic using: `"pip install pydantic"` and `"pip3 install pydantic"`. 
However, after installing it I get the error message:
```
"Error: cannot import name 'Self' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)
Python version 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0]"
```
When I run "pyp list" I get:
```
$ pip3 list | grep typing
typing 3.7.4.3
typing_extensions 4.12.2
```
I completed the installation process indicated in the README, all the other functions work fine except this one. Could you tell me if there is something I am missing or any kind of help you can give me?
If you need more information I can send it to you.
Thanks

Steffen-W, 2024-08-20:
I had the problem myself and it is more or less due to the fact that a necessary feature only comes with python 3.12. But it also works with python 3.10. But it also works with python 3.10. Are you on Linux? 

Try the following: 
sudo apt update 
sudo apt install python3-typing-extensions 

and otherwise 
pip3 install --upgrade typing_extensions

Alviunta, 2024-08-20:
Yes, I'm using Ubuntu 22.04.4 LTS.
I tried both and neither of them work, should I try to upgrade python to 3.12?

I also looked at https://github.com/uPesy/easyeda2kicad.py/blob/master/requirements.txt and it says that I should use at least pydantic 2.0.0.

So I tried to force this version ( `pip install pydantic==2.0.0` ), however now the error changes to:
```
Error: cannot import name 'Required' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)
Python version 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0]
```

sbalula, 2024-09-03:
Got this error (`No module named 'pydantic')` in MacOS 14.6.1, with KiCad 9.0 installed using brew.
Note that KiCad in MacOS uses its own python. For me that's at `/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/`.

So running `/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/pip3.9 --install pydantic` solved the issue.

Thanks @Steffen-W for this very helpful tool!

Steffen-W, 2024-09-03:
Hi @sbalula,
 I'm very pleased that you were able to solve the error. Did you use kicad 8.0.4?
Unfortunately, I am currently too busy to solve the problem in general. Do you have any idea how I could solve the problem within the plugin? Unfortunately, the Python package has also been running on a few of my test devices. Does the plugin now work completely as you would expect?

JoHo94, 2024-09-14:
> Got this error (`No module named 'pydantic')` in MacOS 14.6.1, with KiCad 9.0 installed using brew. Note that KiCad in MacOS uses its own python. For me that's at [...]

Thanks for this tip, this did work for me on windows too.

MKoroschetz, 2024-09-19:
A few days ago I installed KiCad 8.05 on Windows 11 Pro, 24H2, build. 26120.1542 As I am migrating from EasyEDA PRO I am in need to import Symbols and Footprints. to KiCad.
Following the [Readme](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin#) without errors, on first use I got the same error this issue is about.
I found python 3.11 installed by KiCad 8.05 in the "C:\Program Files\KiCad\8.0\bin\Scripts" folder and following the solution suggested by [JoHo94](https://github.com/JoHo94) i was able to solve the problem and make the plugin work as expected.

Thank you [@Steffen-W](https://github.com/Steffen-W) for this very convenient and useful tool, and to [JoHo94](https://github.com/JoHo94) for sharing his finding.

Steffen-W, 2024-09-25:
I speculate that a changeover to Python 3.12 will take place at some point, then the problem will no longer exist. Do any of you have an idea how I could solve the problem within the plugin?

BenRoe, 2024-10-07:
> Got this error (`No module named 'pydantic')` in MacOS 14.6.1, with KiCad 9.0 installed using brew. Note that KiCad in MacOS uses its own python. For me that's at [...]

I am on Mac OSX 15 with M2, Kicad 8.05 and had to change the command to `/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/bin/pip3.9 install pydantic`

Many thanks for the hint.

snenyl, 2025-02-14:
I got KiCAD installed using Flatpak and this worked for me:

**Steps to Install 'pydantic' in KiCad's Flatpak Environment:**

1. **Access the KiCad Flatpak Environment:**
   ```bash
   flatpak run --command=sh org.kicad.KiCad
   ```
   This command opens a shell within the KiCad Flatpak environment.

2. **Install 'pydantic' Using Pip:**
   Within the Flatpak shell, execute:
   ```bash
   pip install pydantic
   ```
   This installs the 'pydantic' module in the Flatpak's Python environment.

3. **Verify the Installation:**
   After installation, you can verify that 'pydantic' is installed by running:
   ```bash
   pip show pydantic
   ```
   This should display information about the installed 'pydantic' package.

4. **Exit the Flatpak Shell:**
   Once the installation is complete, exit the Flatpak shell:
   ```bash
   exit
   ```

5. **Restart KiCad:**
   After performing these steps, restart KiCad to ensure that the changes take effect.

Steffen-W, 2025-03-02:
Can you try the new version [2025.03.01](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.03.01) and see if the problem has been solved?

Steffen-W, 2025-03-04:
The problem should now be solved. If it still exists after the plugin update, please open the issue again.

closed 2025-03-04 by Steffen-W