Steffen-W/Import-LIB-KiCad-Plugin #25: Error: list index out of range

frobgm, 2024-09-09:
Dear Steffen

I try yours plugin but have a little problem, I downloaded from the mouser website lib, and try to import.
whatever I set, I get the same error message  : Error: list index out of range.
[image]
what could be the problem?
Thanks in advance for your help

Steffen-W, 2024-09-09:
Hi @frobgm,

I have never actually had this error. I was able to find the reason and it's not me.

You must correct the following file:
`LIB_TLP5772_D4-TP,E.zip/TLP5772_D4-TP,E/KiCad/TLP5772_D4-TP_E.kicad_sym`
The file is incorrect. Two more brackets are opened than closed. Add two closed brackets  `))` to the end and the import process will not give an error. Depending on your system, you may have to unpack the zip directory, adapt the file and then pack it in the same way. Maybe I will let the plugin output a better error. But I'm more worried about why the format is faulty.

Please let me know if it works for you without errors.

Steffen-W, 2024-09-09:
I am attaching the faulty file [LIB_TLP5772_D4-TP,E.zip](https://github.com/user-attachments/files/16937095/LIB_TLP5772_D4-TP.E.zip) for documentation purposes.
https://www.mouser.de/ProductDetail/Toshiba/TLP5772D4-TPE?qs=6ZHl3BmPMW2Vngmf%252BMnurg%3D%3D

frobgm, 2024-09-10:
Dear Steffen

Now it works fine, that was really the error
Thanks for your help!

Do you have to add the library manually each time, or does your program just expand it after one addition?

Steffen-W, 2024-09-10:
Nice

The program can add the settings automatically. The first time, however, a restart of KiCad is necessary, as the setting can only be loaded at startup. However, you must also select the Auto function in the plugin.

frobgm, 2024-09-11:
Hello Steffen
my friends have also looked into the issue but came to a conclusion later than you
it looks like the mouser part number is wrong in the sym file

(property "Datasheet" "https://componentsearchengine.com/Datasheets/1/TLP5772(D4-TP,E.pdf" (at 36.83 -194.92 0)
      (effects (font (size 1.27 1.27)) (justify left top) hide)
    )

(property "Manufacturer_Part_Number" "TLP5772(D4-TP,E" (at 36.83 -794.92 0)
      (effects (font (size 1.27 1.27)) (justify left top) hide)
    )

and indeed the end of file bracket does solve the issue, but as you can see the error elsewhere is causing this

Translated with DeepL.com (free version)

Steffen-W, 2024-09-11:
Hi @frobgm,
 ok, unfortunately I didn't recognize that right away. Your colleague is right and I have made a mistake. But I have to say that the name is very cryptic.

frobgm, 2024-09-11:
Hi Steffen
Your solution worked, so I wouldn't say it was a mistake.
With error generation, you can only blame the mouser.

frobgm, 2024-09-15:
Hi Steffen
sorry to write again
this one is also wrong, but I didn't find a bracket error in this one, it still runs on import
https://hu.mouser.com/ProductDetail/Bourns/SRN4012BTA-1R2M?qs=VJzv269c%252BPYg2%2FRmAQ%252BNBg%3D%3D
Thank you for checking it

Steffen-W, 2024-09-15:
The file is correct. The only thing that can happen is that the import was previously incorrect with "TLP5772(D4-TP,E ". The added brackets have caused the process to run through, but are unfortunately wrong. 

```
Import: /home/steffen/Downloads/LIB_SRN4012BTA-1R2M.zip
Identify Samacsys
Import kicad_sym
Import footprint
OK
```

I need more information to be able to help. Please always send the issue in text form, this will make my search much easier.

Steffen-W, 2025-01-21:
@frobgm  
I would be happy if you could try out the current version on your site. I will publish it in parallel, but I can also withdraw it quickly if there are any errors.
[releases 2025.01.21](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.01.21)

closed 2025-01-21 by Steffen-W