Steffen-W/Import-LIB-KiCad-Plugin #26: Default ports to "passive" instead of "unspecified"

Riesi-Inmox, 2024-09-18:
When importing some components their IOs are set to "unspecified" which generates a lot of ERC errors.
It would probably make more sense to set ports to "passive" if the type is not clear.

For example this header will all be "unspecified":
https://www.digikey.at/de/models/1955705

Another issue is that the 3d model from the zip isn't assigned. I can create a second issue if you want.

Thanks for the great plugin!
Kind regards Riesi!

Steffen-W, 2024-09-18:
Hi, I can fully understand your statement. Regarding the 3D file, the error is clearly with DigiKey. Have a look at the downloaded zip file and you will see that it is not there. But you can also download it directly from the provider, note the tick mark next to the 3d files.

[image]

I can also understand the part about the ports. However, I have decided not to change the port definitions. The components are imported without any changes. 

If I can help you with anything else, just write. I am shooting the issue as I do not recognise any errors.

closed 2024-09-18 by Steffen-W

Riesi-Inmox, 2024-09-20:
Interesting that you dont get an STL. For me the zip has the 3d model:
[image]

Steffen-W, 2024-09-20:
Interesting, I did not have the file. The problem that is probably the case here is that I do not support digikey per se. For the supported platforms, I try to find out which platform it is via the data structure. So if you download it from the direct provider and not via digikey, it will work. However, I can adapt this so that digikey is also possible without any problems.