Steffen-W/Import-LIB-KiCad-Plugin #27: Error loading library

nutz88, 2024-09-24:
Symbol library 'Snapeda' failed to load.
Expecting '(' in
'C:\Users\xxx\Documents\KiCad\8.0\3rdparty\Snapeda.kicad_sym' line 2, offset 10.

nutz88, 2024-09-24:
[Snapeda.kicad_sym.txt](https://github.com/user-attachments/files/17120239/Snapeda.kicad_sym.txt)

Steffen-W, 2024-09-25:
Hi @nutz88 ,
I need more information about what happened before. It seems to me that you have written manually in the file.

https://www.snapeda.com/parts/RXEF050/Littelfuse/view-part/?welcome=home&ref=dk&t=RXEF050&con_ref=None

I can't find the symbol on snapeda either. In any case, the file is not correct.

Steffen-W, 2024-09-25:
Change the first line, from
```
(kicad_symbol_lib
(symbol "RXEF050"
		(pin_names
			(offset 1.016)
		)
```
to:
```
(kicad_symbol_lib (version 20211014) (generator kicad_symbol_editor)
(symbol "RXEF050"
		(pin_names
			(offset 1.016)
		)
```

Can you reproduce the error somehow? Unfortunately, I still have no idea how the error occurred. Most likely I would say that “RXEF050” was not correct. @nutz88 where here did you get the symbol from?

nutz88, 2024-09-26:
Hi,
I got the symbol from Octopart, but dont remember from what service.
Was able to reproduce it if I deleted the library file in the 3rdparty directory and let the impartgui create them again. It was created the same every time.
I also tried to remove this symbol "RXEF050", but the file was created with the same format, but another component was first in the file instead.
But when i tried again to reproduce it just now, I did not get the error, and the file is formated like you suggest.
What I also did different now was to move all my zip files from my Downloads directory and just do them again one by one to see when the problem happend, but it never did now.
So looks like it might have been just this once

closed 2024-09-26 by nutz88