Steffen-W/Import-LIB-KiCad-Plugin #29: LCSC Part import stop working

MatheusOCorrea, 2024-09-26:
The mensage above appears:

"Try to import EeasyEDA /  LCSC Part# : C5123977
Use overwrite option to update the older symbol
Use overwrite option to replace the older footprint
Created 3D model SOT-23-6_L2.9-W1.6-H1.5-LS2.8-P0.95 (wrl,step)"

But the part does not appear in the library, was working but suddenly stopped.
3D model is create in the library \EasyEDA.3dshapes.

MatheusOCorrea, 2024-09-26:
I found the issue, in EasyEDA.kicad_syn affter six symbols the one bracket close header "(kicad_symbol_lib", after delete the extra bracket, all components added appears.

Steffen-W, 2024-09-26:
```
Try to import EeasyEDA /  LCSC Part# : C476766
Created Kicad symbol C0805B104K500NT
Created Kicad footprint C0805
Created 3D model C0805_L2.0-W1.3-H1.3 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C136141
Created Kicad symbol C4532X5R1E106M250KA
Created Kicad footprint C1812
Created 3D model C1812_L4.5-W3.2-H2.6 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C5123977
Use overwrite option to update the older symbol
Use overwrite option to replace the older footprint
Created 3D model SOT-23-6_L2.9-W1.6-H1.5-LS2.8-P0.95 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C1674
Created Kicad symbol 0603CG5R6C500NT
Use overwrite option to replace the older footprint
Created 3D model C0603_L1.6-W0.8-H0.8 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C1674
Created Kicad symbol 0603CG5R6C500NT
Created Kicad footprint C0603
Created 3D model C0603_L1.6-W0.8-H0.8 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C5123977
Created Kicad symbol LGS63032
Created Kicad footprint SOT-23-6_L2.9-W1.6-P0.95-LS2.8-BR
Created 3D model SOT-23-6_L2.9-W1.6-H1.5-LS2.8-P0.95 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C174766
Created Kicad symbol CR1812F43KE04
Created Kicad footprint R1812
Created 3D model R1812_L4.5-W3.2-H0.6 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C5123977
Created Kicad symbol LGS63032
Created Kicad footprint SOT-23-6_L2.9-W1.6-P0.95-LS2.8-BR
Created 3D model SOT-23-6_L2.9-W1.6-H1.5-LS2.8-P0.95 (wrl,step)
```

I imported the following components for the test:
C23869, C5123977, C2040, C307478, C60474, C476766, C136141, C1674, C174766

I was unable to reproduce the error. However, the components were not overwritten even if I wanted to.

For your information. I imported the plugin [uPesy/easyeda2kicad.py](https://github.com/uPesy/easyeda2kicad.py) and did not write it myself. Only the GUI comes from me. But if the error was caused by me, I will take care of it. Can you reproduce the error somehow so that I can debug it?

I have not found an [issue](https://github.com/uPesy/easyeda2kicad.py/issues?q=)  either.

MatheusOCorrea, 2024-09-26:
Occurs many times last night, but now the error stopped. 
But what happens is this bracket in line 3740 in the EasyEDA.kicad_sym file.

[EasyEDA.kicad_sym.txt](https://github.com/user-attachments/files/17152515/EasyEDA.kicad_sym.txt)

Steffen-W, 2024-09-26:
Very quirky. If it is at all possible for you to reproduce the error, I would be very pleased. I'll look through the code when I get a chance and try to find the source of the error. But I'm assuming that it's somewhere in [uPesy/easyeda2kicad.py](https://github.com/uPesy/easyeda2kicad.py) in any case.

Steffen-W, 2025-01-21:
I'm closing this here as I'm putting the latest version in the plugin and hope the bug doesn't come up again.

closed 2025-01-21 by Steffen-W