Steffen-W/Import-LIB-KiCad-Plugin #31: Plugin fails to load in kicad with an import error

techy-robot, 2024-12-31:
I'm running the latest kicad version, the 8.0.7 build from Ubuntu's 24 package repo, and I'm getting this error:
```
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/asher/.local/share/kicad/8.0/3rdparty/plugins/com_github_Steffen-W_impartGUI/__init__.py", line 9, in <module>
    from .impart_action import ActionImpartPlugin
  File "/home/asher/.local/share/kicad/8.0/3rdparty/plugins/com_github_Steffen-W_impartGUI/impart_action.py", line 136, in <module>
    backend_h = impart_backend()
                ^^^^^^^^^^^^^^^^
  File "/home/asher/.local/share/kicad/8.0/3rdparty/plugins/com_github_Steffen-W_impartGUI/impart_action.py", line 63, in __init__
    self.config = config_handler(path2config)
                  ^^^^^^^^^^^^^^
NameError: name 'config_handler' is not defined
```

I'm not sure why it can't find the import

Steffen-W, 2024-12-31:
Hi @techy-robot,

does the error also occur if you deactivate all other pluins?
Did the plugin work for you in an older version of kicad?

Steffen-W, 2024-12-31:
The error puzzles me a bit, because at the beginning the import apparently did not output any errors.
```python
try:
    if __name__ == "__main__":
        from impart_gui import impartGUI
        from KiCadImport import import_lib
        from impart_helper_func import filehandler, config_handler, KiCad_Settings
        from impart_migration import find_old_lib_files, convert_lib_list
    else:
        # relative import is required in kicad
        from .impart_gui import impartGUI
        from .KiCadImport import import_lib
        from .impart_helper_func import filehandler, config_handler, KiCad_Settings
        from .impart_migration import find_old_lib_files, convert_lib_list
except Exception as e:
    print(traceback.format_exc())
```

techy-robot, 2024-12-31:
I haven't used this in older versions yet, I just recently installed it. The error does not occur when other plugins are disabled, so I'll have to test and narrow it down to which one is conflicting.

techy-robot, 2024-12-31:
Okay, the Impart plugin no longer works when the Kicad Parasitics plugin is enabled in the plugin directory. The parasitics plugin is a bit out of date with kicad main, so there could be some conflict. https://github.com/Steffen-W/KiCad-Parasitics

Steffen-W, 2025-01-01:
Hi, unfortunately I can't reproduce the error and I don't know exactly how it comes about. The other plugin also comes from me. Unfortunately I have not really worked on this one, but I can make the necessary changes. If you have an idea where the problem is please write to me.

frobgm, 2025-01-09:
Hi Steffen
Again, I am looking for you with an import error.
I'm attaching the file that the plugin you requested creates.
In fact both the file and the library show the part and the leg assignment, but in the program I don't see only the first part in the list.
Thanks in advance for your help!
[Samacsys.zip](https://github.com/user-attachments/files/18359642/Samacsys.zip)

Steffen-W, 2025-01-09:
https://componentsearchengine.com/part-view/TLP5772(D4-TP%2CE/Toshiba

Hi @techy-robot 
Same like #25.

Problem: 

```python
        def extract_symbol_section(input_text):
            start_index = input_text.find("(symbol")  # Search for "(symbol"
            if start_index == -1:
                return None
            open_brackets = 1
            end_index = start_index + len("(symbol")
            for i in range(start_index + len("(symbol"), len(input_text)):
                if input_text[i] == "(":
                    open_brackets += 1
                elif input_text[i] == ")":
                    open_brackets -= 1
                    if open_brackets == 0:
                        end_index = i + 1
                        break
            symbol_section = input_text[start_index:end_index]
            return symbol_section, start_index, end_index
```
The url contains a kalmmer that my primitive paser cannot cope with. A suitable paser is already in the project but would have to be used in the problematic function. The problem is the bracket in the URL.

Solution is to use: https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/s_expression_parse/__init__.py

If you have the time, I would be happy if you could work out a solution. Otherwise, delete the bracket in the file in the url and the import will work.

frobgm, 2025-01-09:
Thank you for your reply!
First of all I replaced the linked __init__.py but not even the plugin loaded in the menu bar, and it didn't appear in the pcb designer under plugins either, I put the old one back.
And now I couldn't even get the bracket to breathe life into it.
only the first component appears, nothing else :(

closed 2025-01-21 by Steffen-W