Steffen-W/Import-LIB-KiCad-Plugin #33: Does not work with version 8.0.8-RC1-3

webtronico, 2025-01-09:
Does not work with version 8.0.8-RC1-3
It installs normally and is present in the plugins directory, but it does not appear in the PCB Editor bar, as in the PCB Editor -> Tools -> External Plugins menu list.

[image]
[image]
[image]

Steffen-W, 2025-01-10:
Hi @webtronico, Can you start kicad via the terminal? Then I can get more information about the problem. Simply go to the location where kicad is installed and start kicad.exe via the terminal.

nickoe, 2025-01-10:
@Steffen-W  Related to this, I am trying on windows with kicad 9.0 and getting a similar issue:

```
C:\Program Files\KiCad\9.0\bin>An exception occurred during import C:\Users\noestergaard\Documents\KiCad\9.0\3rdparty\plugins\com_github_Steffen-W_impartGUI\config.ini
```

There is no config.ini in that location.

impart version is 2024.08.17 installed via the PCM. 

Version: 9.0.0-rc1-105-gfb8d16439b, release build

webtronico, 2025-01-10:
Hi, could this be the information you are requesting? (see screenshot)
[image]
[image]

nickoe, 2025-01-14:
Hi @Steffen-W What do we need to do to resolve this? Do we need to put in a valid config file in that location as a workaround for that try except block? What should it look like, if we need to test it?

Steffen-W, 2025-01-16:
Hi @nickoe, sorry for the late reply. Just add [config.ini](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/config.ini) and tell me the error afterwards.

[impart_helper_func.py#L47-L51](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/impart_helper_func.py#L47-L51)
``` python
        try:
            self.config.read(self.config_path)
            self.config["config"]["SRC_PATH"]  # only for check
            self.config["config"]["DEST_PATH"]  # only for check
            self.config_is_set = True
```

I would hope that it doesn't matter whether the file exists or not.

Steffen-W, 2025-01-16:
You could put large parts of the code in the following query and send me the error. I don't think that the missing file is the problem, rather it shows that the plugin is basically starting. At least until  [impart_action.py#L63C1-L63C50](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/impart_action.py#L63C1-L63C50)

```python
try:

    code .....

except Exception as e:
    print(traceback.format_exc())
```

Try the following:
[impart_action.py#L61-L74](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/impart_action.py#L61-L74)

```python
    def __init__(self):
        try:
            path2config = os.path.join(os.path.dirname(__file__), "config.ini")
            self.config = config_handler(path2config)
            path_seting = pcbnew.SETTINGS_MANAGER().GetUserSettingsPath()
            self.KiCad_Settings = KiCad_Settings(path_seting)
            self.runThread = False
            self.autoImport = False
            self.overwriteImport = False
            self.import_old_format = False
            self.autoLib = False
            self.folderhandler = filehandler(".")
            self.print_buffer = ""
            self.importer = import_lib()
            self.importer.print = self.print2buffer
        except Exception as e:
            print(traceback.format_exc())
```

mairas, 2025-01-20:
Even though my issue is about 9.0 RC2, I suspect the root cause might be at least partially the same, so I am reporting it here.

My symptoms are the same: the plugin installs fine but isn't visible in the toolbar or in the menu. In Settings -> PCB Editor -> Plugins (not sure if that even existed in 8.0), there's a button to show the plugin errors, and I'm getting the following stack trace there:

```
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/mairas/Documents/KiCad/9.0/3rdparty/plugins/com_github_Steffen-W_impartGUI/__init__.py", line 9, in <module>
    from .impart_action import ActionImpartPlugin
  File "/Users/mairas/Documents/KiCad/9.0/3rdparty/plugins/com_github_Steffen-W_impartGUI/impart_action.py", line 136, in <module>
    backend_h = impart_backend()
  File "/Users/mairas/Documents/KiCad/9.0/3rdparty/plugins/com_github_Steffen-W_impartGUI/impart_action.py", line 80, in __init__
    if version_to_tuple(pcbnew.Version()) < version_to_tuple(minVersion):
  File "/Users/mairas/Documents/KiCad/9.0/3rdparty/plugins/com_github_Steffen-W_impartGUI/impart_action.py", line 77, in version_to_tuple
    return tuple(map(int, version_str.split(".")))
ValueError: invalid literal for int() with base 10: '0-rc2'
```

In essence, it seems the version number isn't parsed properly.

In any case, thank you for the great plugin! It's a huge time-saver!

Steffen-W, 2025-01-20:
Hi @mairas,
If you delete the following part [plugins/impart_action.py#L76-L83](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/6680941eff71bc6a2293ef6add1dc26cd76828a5/plugins/impart_action.py#L76-L83), will the plugin work? Please also print(pcbnew.Version()). Unfortunately I don't have the time to make the plugin fully functional again. I'll try today. If you like to fix the error yourself, I'm happy too. Or does #32 already solve your problem?

mairas, 2025-01-20:
Yes, actually I had done exactly that on my own.

PR #32 probably does fix this particular issue, but I recommend using e.g. the [semver](https://github.com/python-semver/python-semver) library for comparing version numbers.

nickoe, 2025-01-21:
> Hi [@nickoe](https://github.com/nickoe), sorry for the late reply. Just add [config.ini](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/config.ini) and tell me the error afterwards.
> 
> [impart_helper_func.py#L47-L51](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/blob/master/plugins/impart_helper_func.py#L47-L51)
> 
>         try:
>             self.config.read(self.config_path)
>             self.config["config"]["SRC_PATH"]  # only for check
>             self.config["config"]["DEST_PATH"]  # only for check
>             self.config_is_set = True
> I would hope that it doesn't matter whether the file exists or not.

This made no difference, in that it does not load the plugin enough, but I don't see that exception though, and I added the "empty" config.ini as you mentioned:
```
type C:\Users\user\Documents\KiCad\9.0\3rdparty\plugins\com_github_Steffen-W_impartGUI\config.ini
[config]
src_path =
dest_path =
```

From the console in pcbnew:
```
import pcbnew
print(pcbnew.Version())
9.0.0-rc1
```

I can confirm that the fix in https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/pull/32 works!

closed 2025-01-21 by Steffen-W

Steffen-W, 2025-01-21:
@webtronico, @nickoe and @mairas 
I would be happy if you could try out the current version on your site. I will publish it in parallel, but I can also withdraw it quickly if there are any errors.
[releases 2025.01.21](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.01.21)

nickoe, 2025-01-21:
It appears to work when using the zip from the release tag you are linking in
https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.01.21

I think the pcm repo can be bumped as well :)