Steffen-W/Import-LIB-KiCad-Plugin #36: Call script from command line

nmammeri, 2025-02-08:
Thanks for this excellent plugin.

I'm wondering if there is way to call the python script directly and not relying the PCM GUI? 

I'm thinking of something like:
`import-lib-kicad.py --dowanlod-folder ~/Dowloads --lib-folder ~/Kicad`

I can call this script whenever I download a new component.

Many thanks

Steffen-W, 2025-02-09:
Hi, of course it is possible. I just haven't seen any added value in it so far. You would want to start it without a GUI, wouldn't you? It's not difficult to implement, but I've never needed it myself.

Steffen-W, 2025-02-09:
There is a change to the API in kicad 9. In the course of this, I wanted to adapt the plugin anyway. Let's see when I find the time. The new API also allows you to access kicad from outside without any problems. This means that plugins can also be used normally from outside.

Steffen-W, 2025-02-09:
Hi @nmammeri, did you mean something like that? Please test it.

Example: 
```Bash
python plugins/KiCadImport.py --download-folder Demo/libs --lib-folder import_test
```

closed 2025-02-09 by Steffen-W

nmammeri, 2025-02-09:
Yes exactly. This way it can also be used to create project libraries instead of just global libraries.

By the way, with the current version I managed to use the plugin to create a project library. It works great except that the 3d model path in the footprint expects Kicad3dParty environment variable. I had to go through the components manually and change the paths. It would be great if the 3d model path is relative to the footprint.

Thanks

Steffen-W, 2025-02-09:
I can give you the option to pass the entry manually. Would you like to do it without a variable and specify it as a relative path?

Steffen-W, 2025-02-09:
Hi @nmammeri,

I have now expanded this. You can now pass anything, including the project-specific variable or another path such as `${KIPRJMOD}/lib`.

```bash
python plugins/KiCadImport.py --download-folder Demo/libs --lib-folder import_test --path-variable '${KIPRJMOD}' --overwrite-if-exists
```

Steffen-W, 2025-02-09:
Perhaps I will also extend the gui so that the import can also be made project-based.