Steffen-W/Import-LIB-KiCad-Plugin #37: Problem with pathes containing spaces?

callimero, 2025-02-21:
I get this:

Identify Snapeda
[Errno 22] Invalid argument: 'C:\\Users\\Carsten Wartmann\\Documents\\KiCad\\9.0\\3rdparty\t\\Snapeda.kicad_sym~'
Error: [Errno 22] Invalid argument: 'C:\\Users\\Carsten Wartmann\\Documents\\KiCad\\9.0\\3rdparty\t\\Snapeda.kicad_sym~'
Python version 3.11.5 (main, Jan 27 2025, 07:08:36) [MSC v.1942 64 bit (AMD64)]

When I change to a dir without spaces it is satisfied :)

Steffen-W, 2025-02-21:
Hi @callimero,
Which version of the plugin are you using?
Can you start Kicad via the console in the hope of getting more output?
Does the file ‘C:\Users\Carsten Wartmann\Documents\KiCad\9.0\3rdparty\t\Snapeda.kicad_sym~’ exist?
Does the parent folder exist?
Do you have write permissions in this area?

Steffen-W, 2025-03-02:
@callimero can you send me your config.ini when the problematic configuration is set? It is in the plugin directory.

callimero, 2025-03-03:
I solved it using just a path without spaces.
I will try again with KiCAD 9 soon.

Steffen-W, 2025-03-04:
Unfortunately, I have not been able to reproduce the error. But I assume that the blank character was not the problem but rather \t or something? If necessary, I can open the issue again

closed 2025-03-04 by Steffen-W