Steffen-W/Import-LIB-KiCad-Plugin #38: doesn't import

sirus20x6, 2025-03-01:
arch linux, kicad 9, installed plugin, restarted, added paths from the github readme.md, tried to add part, get the following output:

Try to import EeasyEDA /  LCSC Part# : C5364646
Created Kicad symbol ESP32-C6_C5364646
Created Kicad footprint QFN-40_L5.0-W5.0-P0.40-TL-EP3.3
Error: 'NoneType' object has no attribute 'step'
Python version 3.13.2 (main, Feb  5 2025, 08:05:21) [GCC 14.2.1 20250128]

symbols do not populate in the schematic symbol placer, also only easyeda is available from the dropdown. no other sources.

closed 2025-03-01 by Steffen-W

Steffen-W, 2025-03-01:
Hi @sirus20x6,

The problem was that no 3D file exists on EeasyEDA. I have made a few adjustments. Can you try the new version [2025.03.01](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.03.01) and see if the problem has been solved?

sirus20x6, 2025-03-01:
Thanks! I'll give it a try

sirus20x6, 2025-03-01:
so I now get

```
Try to import EeasyEDA /  LCSC Part# : C5364646
Created Kicad symbol ESP32-C6_C5364646
Created Kicad footprint QFN-40_L5.0-W5.0-P0.40-TL-EP3.3
No 3D model available for this component.
```
but I still don't see anything in the EASYEDA library, and searching for esp32-c6 just brings up the module and not the chip that correlates with the part number I put in

Steffen-W, 2025-03-02:
Was the library created in the specified folder? Does the file that you import into KiCad exist or what is the problem? The component is imported normally for me and I can't find any errors.

sirus20x6, 2025-03-04:
```
Try to import EeasyEDA /  LCSC Part# : C68740
Created Kicad symbol CU3M002_USB A/M 3.0 SMT L=1.9 K
Created Kicad footprint USB-3.0-SMD_CU3M002_USBA-M
Created 3D model USB-3.0-SMD_9P-L12.0-W18.9-H4.5-LS20.8-P1.0 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C68740
Created Kicad symbol CU3M002_USB A/M 3.0 SMT L=1.9 K
Created Kicad footprint USB-3.0-SMD_CU3M002_USBA-M
Created 3D model USB-3.0-SMD_9P-L12.0-W18.9-H4.5-LS20.8-P1.0 (wrl,step)

Try to import EeasyEDA /  LCSC Part# : C68740
Created Kicad symbol CU3M002_USB A/M 3.0 SMT L=1.9 K
Created Kicad footprint USB-3.0-SMD_CU3M002_USBA-M
Created 3D model USB-3.0-SMD_9P-L12.0-W18.9-H4.5-LS20.8-P1.0 (wrl,step)
```

"Was the library created in the specified folder? Does the file that you import into KiCad exist or what is the problem? The component is imported normally for me and I can't find any errors."

no files / folders were created that I can tell, the symbol does not show up in the place symbol menu.

btw I have no idea how to use kicad, I'm coming from altium, but I do have pretty solid linux knowledge.

Steffen-W, 2025-03-04:
Hi @sirus20x6, 
I know some people who switched and then stayed with kicad for the long term. The training is worth it. Please watch the video https://youtu.be/BYIKjCs1qKQ. If the question is still unanswered, just write to me.

sirus20x6, 2025-03-04:
oh I've already watched that one a few dozen times. somehow I was able to get a part to show up in the symbol library, and the footprint editor, but it wont sync from the schematic over to the pcb 

[image] I've added the correct .pretty file but no dice

Steffen-W, 2025-03-04:
easyeda is not EasyEDA

Steffen-W, 2025-03-04:
@sirus20x6  in general. i would only ever activate the last three things in the window on the right but only very rarely the first. just as a general tip.

sirus20x6, 2025-03-04:
ah I see. I though that was just a short name for humans not something that had to match between symbol and footprint. I'm certainly going to miss a few things from altium but at least the schematic and pcb window update a million times faster

commit b0c8b111 (2025-03-01): Fix #38: Handling with missing 3d model