Steffen-W/Import-LIB-KiCad-Plugin #40: Does it support KiCad 9.0？

xiajintaord, 2025-05-21:
After installing this plugin, I successfully imported the component symbols and packaging of LCSC, but I couldn't find the newly imported component symbols when placing them in the schematic interface; I can see the imported file symbols and packaging in the folder, but I couldn't find them when placing them. The paths to the library files are all correct

[image]

[image]

[image]

Steffen-W, 2025-05-21:
Sorry, I don't quite understand the problem yet. Did you also import the footprint library? According to the images, everything looks correct. Everything should run in kicad 9 as in kicad 8. But soon there will be a big change and only kicad 9 will be supported by me.

xiajintaord, 2025-05-22:
> Sorry, I don't quite understand the problem yet. Did you also import the footprint library? According to the images, everything looks correct. Everything should run in kicad 9 as in kicad 8. But soon there will be a big change and only kicad 9 will be supported by me.
My problem is that after successfully importing the component, when placing symbols on the schematic interface, I cannot find the component that was just imported (the symbol library and package library path settings are correct). You can test it in KiCad 9.0 version according to my instructions

Steffen-W, 2025-05-22:
Ok, that could be the case in principle, as the importer does not come directly from me. But it will work in the next version. But can you link the footprint manually or is it missing?

xiajintaord, 2025-05-23:
> Ok, that could be the case in principle, as the importer does not come directly from me. But it will work in the next version. But can you link the footprint manually or is it missing?

yes! I can you link the footprint manually.

[image]

[image]

Look at these two pictures above, the number of symbols and the number of footprints do not match. All footprints were successfully imported, but some symbols were not imported successfully

xiajintaord, 2025-05-23:
I changed the import path today to point to a newly created symbol library and footprint library, and everything is back to normal now.
 I will continue to observe and see if there is another issue where symbols cannot be found after importing components

closed 2025-05-25 by Steffen-W