Steffen-W/Import-LIB-KiCad-Plugin #41: Symbol not creating

Roblocz, 2025-05-25:
Hi so when i do a manual import for example from lcsc everything is good theres no error and in the output it says

Try to import EeasyEDA /  LCSC Part# : C2838500
Created Kicad symbol ESP32-C3
Created Kicad footprint QFN-32_L5.0-W5.0-P0.50-TL-EP3.7
Created 3D model QFN-32_L5.0-W5.0-H0.9-P0.50-TL-EP3.7 (wrl,step)

i already added both symbol and footprint easyeda libraries and it only shows on the footprint library 

Thanks 
Isaias

Roblocz, 2025-05-25:
i just changed name of the easyeda library for the plugin to create another new one and it worked like it should

Steffen-W, 2025-07-19:
https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.07.19

Hi @Roblocz, if you have the time and inclination, I would appreciate it if you could take a look at the latest version to see if your bug is fixed. It is mostly a new implementation, best used with kicad 9.

Roblocz, 2025-07-20:
Hi
Thanks for the heads-up. I’ll test the new version with KiCad 9 and let you know if I find anything.
Thank you i really like the plugin!

Steffen-W, 2025-08-01:
@igsz0,
Everything should now work as desired. I would be delighted if you tried it yourself.
https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.08.01