Steffen-W/Import-LIB-KiCad-Plugin #47: Impart GUI Above version 2025.03.03 Installs But WIll Not Run on my Kicad v9.0.3 Installation

jimmysr, 2025-08-02:
I've been running version 2025.03.03 since it came out with low to no issues. I updated to 2025.08.01 when it became available today.  It installed with no errors or issues. But the icon is not visible on the tool bar in PCBNEW and it is not visible under "Tools>External Plugins". 

I've uninstalled and reinstall several times with the same results. I went back and installed 2025.03.03 and it works just fine.

Note that last week I attempted to install the 2025.07.?? version with the same result as 2025.08.01.

I am not much of a scripter and have little to no experience in Python so I haven't fumbled around other than to very the installation folder and kicad variables are as correct as I can tell.

My email is jim@mayfieldj.com

CarlosPichardo, 2025-08-03:
Same problem

jimmysr, 2025-08-03:
I have the same issue on my MacBook Pro. It is running Kicad v9.0.3. It runs impart gui version 2025.03.03 and below just fine but will not run the 2025.08.01 version.

Steffen-W, 2025-08-03:
Hi, sorry to hear that there are problems, but thank you for creating this issue. The functionality has changed significantly in the current version and hopefully makes many things better. I think the problem is with reloading the dependencies. I had included these in the plugin in the old version, but now I let KiCad reload them. This means that it may take a few seconds for the plugin to appear. Unfortunately, I have only tested the current version extensively on Linux, so there may still be problems with other operating systems. I will try to resolve this as quickly as possible. Please help me by providing information about your KiCad version, starting KiCad in debug mode, and sending me your output. That would be a great help.

If you really want to use the plugin, use version 2025.03.03. This version is secure and should not cause any problems. Subsequent versions behave differently.

---

Here's how to start KiCad with debug output on different operating systems:

## Linux

**Terminal command:**
```bash
export KICAD_ENABLE_WXTRACE=1
export WXTRACE=KICAD_API
kicad
```

**Or as one-liner:**
```bash
KICAD_ENABLE_WXTRACE=1 WXTRACE=KICAD_API kicad
```

## Windows

If necessary, the path to KiCad must be adjusted.

**Command Prompt (cmd):**
```cmd
set KICAD_ENABLE_WXTRACE=1 && set WXTRACE=KICAD_API && "C:\Program Files\KiCad\9.0\bin\kicad.exe"
```

**PowerShell:**
```powershell
$env:KICAD_ENABLE_WXTRACE=1; $env:WXTRACE="KICAD_API"; & "C:\Program Files\KiCad\9.0\bin\kicad.exe"
```

## macOS

**Terminal command:**
```bash
export KICAD_ENABLE_WXTRACE=1
export WXTRACE=KICAD_API
/Applications/KiCad/KiCad.app/Contents/MacOS/kicad
```

**Or as one-liner:**
```bash
KICAD_ENABLE_WXTRACE=1 WXTRACE=KICAD_API /Applications/KiCad/KiCad.app/Contents/MacOS/kicad
```

## Debug Options

You can use different WXTRACE values depending on what you want to debug:
- `WXTRACE=KICAD_API` - for API-related output
- `WXTRACE=KICAD_PYTHON` - for Python plugin debug output
- `WXTRACE=*` - for all available trace messages

Debug output will appear in the console/terminal where you started KiCad.

# How to Get KiCad Version Information

To display detailed KiCad version and build information like you showed, follow these steps:

1. Open KiCad
2. Go to **Help** → **About KiCad**
3. In the About dialog, click on **"Copy Version Info"** button
4. The complete system information will be copied to your clipboard

Steffen-W, 2025-08-03:
[image]
It seems to help to activate the KiCad API. However, this should not be necessary for internal plugins. Please (@jimmysr, @CarlosPichardo) test it anyway to see if it solves the problem. Then I can address the problem directly.

Steffen-W, 2025-08-03:
> <img alt="Image" width="1004" height="515" src="https://private-user-images.githubusercontent.com/52629706/473764973-9e089924-c31d-4a50-a783-c31cb45e80e0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQyMzkxMDMsIm5iZiI6MTc1NDIzODgwMywicGF0aCI6Ii81MjYyOTcwNi80NzM3NjQ5NzMtOWUwODk5MjQtYzMxZC00YTUwLWE3ODMtYzMxY2I0NWU4MGUwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODAzVDE2MzMyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMwMTJmZGZjMTI1YzFkNzljYzFjMGExZjIwMmZmMjI2ZjBkM2IzYmE5YzdlOTU4YjczNmIzNTE0ZDE3NDVjN2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bKSUYvS4bt_i847aWVS5A1RCyZj_2GOvzCrW-9UhWFE"> It seems to help to activate the KiCad API. However, this should not be necessary for internal plugins. Please ([@jimmysr](https://github.com/jimmysr), [@CarlosPichardo](https://github.com/CarlosPichardo)) test it anyway to see if it solves the problem. Then I can address the problem directly.

@jimmysr Have you ever tried this?

Steffen-W, 2025-08-03:
I am currently troubleshooting. The dependencies are probably the problem. They are not really installed for whatever reason. @jimmysr  can look in the plugin directory for a plugin.log file. And also basically whether the plugin exists at all.

Steffen-W, 2025-08-03:
@jimmysr ,

please test the following version. This should have fixed the problem. If not please give me “init.log” “plugin.log”.
[Import-LIB-KiCad-Plugin.zip](https://github.com/user-attachments/files/21567689/Import-LIB-KiCad-Plugin.zip)

jimmysr, 2025-08-04:
> Steffen-W left a comment [ (Steffen-W/Import-LIB-KiCad-Plugin#47) ]( https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/issues/47#issuecomment-3148652146 )

Sorry just got back home. Testing it now.
 
 

Jim Mayfield
 “Democracy is two wolves and a lamb voting on what to have for lunch. Liberty is a well-armed lamb contesting the vote.”

jimmysr, 2025-08-04:
OK. I've never installed from a zip file before so I hope I did it correctly. I uninstalled my working version and performed an Install from file in the plugin manager. Appears there is a 3.0.2 version, no download size, no install size, check mark in compatible, status is stable.  It installed no errors. I restarted Kicad and there is now an icon on the tool bar. I clicked the icon and the plugin screen launched.  I entered the folder where my library is saved. It took it no errors.

Plugin execution appeared normal until I entered an LCSC part # and clicked Manual import. Nothing happens. I existed teh plugin and Kicad and tried one more time. Same result. Th plugin takes my lcsc number, but does not do anything when I click Manual import. I tried checking Overwrite and auto Kicad settings. But nothing.

This is what I see in the plugin windows. Appears normal to me.
Tip: You can drag ZIP files directly into this window!
==================================================

Library Mode: Global Library
Source Directory: C:\Users\jim\Downloads
Destination Directory: F:\MyLibraries\LCSC
==================================================
Tip: You can drag ZIP files directly into this window!
==================================================

Library Mode: Global Library
Source Directory: C:\Users\jim\Downloads
Destination Directory: F:\MyLibraries\LCSC
==================================================

It appears you're getting closer.

Thank you

beniloew, 2025-08-08:
Same issue here

papyDoctor, 2025-08-09:
Same issue on MACOS Kicad 9

sslupsky, 2025-08-11:
Same issue.
```
Application: KiCad arm64 on arm64

Version: 9.0.3, release build

Libraries:
	wxWidgets 3.2.8
	FreeType 2.13.3
	HarfBuzz 10.2.0
	FontConfig 2.15.0
	libcurl/8.7.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.12 nghttp2/1.64.0

Platform: macOS Sequoia Version 15.6 (Build 24G84), 64 bit, Little endian, wxMac
OpenGL: Apple, Apple M2 Max, 2.1 Metal - 89.4

Build Info:
	Date: Jul  7 2025 22:10:04
	wxWidgets: 3.2.8 (wchar_t,wx containers)
	Boost: 1.87.0
	OCC: 7.8.1
	Curl: 8.7.1
	ngspice: 44.2
	Compiler: Clang 16.0.0 with C++ ABI 1002
	KICAD_IPC_API=ON
```

Steffen-W, 2025-08-12:
https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.08.12
Please test it and I will then release it. It should solve the problems, as I have created a fallback solution based on pcbnew that does not use the new IPC API.

jimmysr, 2025-08-12:
Downloaded and testing now on my Windows 10 environment, Will let you know shortly.

jimmysr, 2025-08-12:
I've completed initial testing on my Windows 10, Kicad v9.0.3 and it appears you've got a working plugin. I imported over 35 components from lcsc, and another dozen from Mouser & Snapeda. So far Ive had no issues in the Windows 10 environment. Going to the MabBook Pro now and will let you know.

Thank you for your efforts.

Steffen-W, 2025-08-20:
@SalmoF can you try this: [Import-LIB-KiCad-Plugin.zip](https://github.com/user-attachments/files/21904300/Import-LIB-KiCad-Plugin.zip)

jimmysr, 2025-08-21:
I could not get it to install on my MACBOOK Pro. It stated it would install a virtual environment.  It took about 3 minutes and threw an error stating if failed to create a virtual environment. So I tried the Kicad API but it also failed to work.

jimmysr, 2025-08-21:
[plugin_fallback.log](https://github.com/user-attachments/files/21916470/plugin_fallback.log)

jimmysr, 2025-08-21:
[plugin_fallback.log](https://github.com/user-attachments/files/21927522/plugin_fallback.log)

I installed and am sorry to say I experienced the same results.

Steffen-W, 2025-08-22:
That's a shame. How do you usually create a virtual environment? I could change it for you so that it doesn't crash. Unfortunately, I'm not very familiar with macOS, but I'll look for a solution later today.

jimmysr, 2025-08-22:
I am able to work on my Windows 10 machine. My Mac is my on the road machine. I use the Windows machine 90% of the. So I am working just fine on the latest version you've provided. And I can always fallback to the 20250303 version on the MAC.

I was just wanting to assist in helping to get the MAC version working. Although I am not really a MAC person I can usually stumble around.

Steffen-W, 2025-08-22:
Hi @jimmysr, I appreciate any help. I don't have direct access to a Mac for testing, so it would be very helpful if someone could do that. I've done some reading and it seems to me that a standard Python installation is not required on a Mac. Furthermore, I can't really use the Python instance that KiCad comes with. I can execute Python code, but the fallback solution requires a virtual environment, which, as far as I understand, I cannot create with it. However, I can add to the plugin that it searches for a Python installation and uses it. Otherwise, I don't know what else to do. But what should work is the variant via the new API. If you could activate it on your end and let me know if it works, I would be very happy.

jimmysr, 2025-08-24:
Here is a screenshot & logfile of when I ran impart plugin. My Python3 version SSL is way out of date. I am going to update my python which will update the SSL. Should have done it a long while ago.

[image]

[plugin_fallback.log](https://github.com/user-attachments/files/21959281/plugin_fallback.log)

SiTLOOMS, 2025-08-25:
> [@jimmysr](https://github.com/jimmysr) can you try https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.08.24 ?

Fixed the issue for me. 

Version: 9.0.4, release build
Libraries:
	wxWidgets 3.2.8
	FreeType 2.13.3
	HarfBuzz 10.2.0
	FontConfig 2.15.0
	libcurl/8.13.0-DEV Schannel zlib/1.3.1
Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: NVIDIA Corporation, NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2, 4.6.0 NVIDIA 566.07
Build Info:
	Date: Aug 18 2025 00:41:45
	wxWidgets: 3.2.8 (wchar_t,wx containers)
	Boost: 1.88.0
	OCC: 7.9.1
	Curl: 8.13.0-DEV
	ngspice: 44
	Compiler: Visual C++ 1942 without C++ ABI
	KICAD_IPC_API=ON
Locale: 
	Lang: en_GB
	Enc: UTF-8
	Num: 1,234.5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)

Steffen-W, 2025-08-25:
https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/releases/tag/2025.08.25

I have now solved the problem by eliminating the cause of the dependencies. Venv is no longer required in fallback mode, so the error should be permanently resolved for everyone. Please test the latest version and create an issue if there are any errors. I will close this issue, but it can still be answered, of course.

closed 2025-08-25 by Steffen-W

Steffen-W, 2025-08-26:
I have moved it to a new issue, as it should not be a general problem. The import works quite well, but the frame symbol is missing. Otherwise, I cannot identify any general problems at this point.
```
Starting import for EasyEDA/LCSC component: C112537
Added symbol: LM317LIPK
Created footprint: SOT-89_L4.5-W2.5-P1.50-LS4.2-BR.kicad_mod
Created 3D model (WRL): SOT-89-3_L4.3-W2.5-H1.6-LS4.1-P1.50.wrl
Created 3D model (STEP): SOT-89-3_L4.3-W2.5-H1.6-LS4.1-P1.50.step
EasyEDA import completed successfully! (4 files created)

Starting import for EasyEDA/LCSC component: C97522
Added symbol: W25Q256JVEIQTR
Created footprint: WSON-8_L8.0-W6.10-P1.27-BL-EP.kicad_mod
Created 3D model (WRL): WSON-8_L8.0-W6.0-H0.78-P1.27.wrl
Created 3D model (STEP): WSON-8_L8.0-W6.0-H0.78-P1.27.step
EasyEDA import completed successfully! (4 files created)
```

[image]
[image]
[image]
[image]