Steffen-W/Import-LIB-KiCad-Plugin #48: Updating an imported symbol prevents additional symbols from appearing in import library

UsableLoki8221, 2025-08-13:
I have impart GUI set to import to a global folder location.  Upon importing a symbol and then doing pin allocations to electrical types or adjusting symbol shape then any new parts imported after the first part's symbol is changed will not appear in the symbol or footprint libraries.  The files do get downloaded successfully into my global folder but they won't appear in the symbol library or footprint library anymore where they previously did.

UsableLoki8221, 2025-08-14:
[plugin.log](https://github.com/user-attachments/files/21775663/plugin.log)

Here is a screen recording I made of the issue (no error message btw)
https://drive.google.com/file/d/1qbHzDZsuXAjr82INWPPwrmg-Ml-DlfzX/view?usp=drive_link

I show how there's two LCSC number not currently populated in my library.  I import one, then edit the symbol to make pin assignments, save, show that it imports then attempt to import the second LCSC number and show how it does not appear.  I have been creating new save folder locations to restore the Impart GUI capability and setting the old broken one to EasyEDAx (where x is just a number to change the string).

UsableLoki8221, 2025-08-14:
I uploaded the video to youtube in case the drive link gives issues

https://youtu.be/JTaDQfLUQiI

Steffen-W, 2025-08-14:
@UsableLoki8221, can you upload a damaged lib file? That would speed up the search considerably.

Steffen-W, 2025-08-14:
The error is very easy to see. Thank you very much. The closed bracket is before the last component added. The problem will probably be temporarily fixed if you delete the closed bracket before “ (symbol ”XL-5050RGBC-WS2812B“”). Unfortunately, I don't have time to fix the error today, but I'll probably be able to do it tomorrow. Thank you very much for your support.

```
2025-08-14 10:58:14,191 INFO [root:impart_action.py:720]: Successfully imported EasyEDA component C2843785
```
```
(kicad_symbol_lib
	(version 20241209)
	(generator "kicad_symbol_editor")
	(generator_version "9.0")
	(symbol "ESP32-S3-MINI-1-N8"
		(exclude_from_sim no)
		(in_bom yes)
		(on_board yes)
		(property "Reference" "U"
			(at 0 46.99 0)
			(effects
				(font
					(size 1.27 1.27)
				)
			)
		)

.....
				)
				(number "7"
					(effects
						(font
							(size 1.27 1.27)
						)
					)
				)
			)
		)
		(embedded_fonts no)
	)
)
  (symbol "XL-5050RGBC-WS2812B"
    (in_bom yes)
    (on_board yes)
    (property
      "Reference"
      "LED"
      (id 0)
      (at 0 6.35 0)
      (effects (font (size 1.27 1.27) ) )
    )
    (property
      "Value"
      "XL-5050RGBC-WS2812B"
      (id 1)
      (at 0 -6.35 0)
      (effects (font (size 1.27 1.27) ) )
    )
    (property
      "Footprint"
      "EasyEDA:LED-SMD_4P-L5.0-W5.0-BL_XL-5050RGBC"
      (id 2)
      (at 0 -8.89 0)
      (effects (font (size 1.27 1.27) ) hide)
    )
    (property
      "LCSC Part"
      "C2843785"
      (id 5)
      (at 0 -11.43 0)
      (effects (font (size 1.27 1.27) ) hide)
    )
    (symbol "XL-5050RGBC-WS2812B_0_1"
      (rectangle
        (start -6.35 3.81)
        (end 6.35 -3.81)
        (stroke (width 0) (type default) (color 0 0 0 0))
        (fill (type background))
      )
....
      (pin unspecified line
        (at 8.89 1.27 180)
        (length 2.54)
        (name "DIN" (effects (font (size 1.27 1.27))))
        (number "4" (effects (font (size 1.27 1.27))))
      )
    )
  )
)
```

UsableLoki8221, 2025-08-22:
@Steffen-W Seems like it works!  I only checked updating a symbol globally- I haven't tried updating a global footprint, do you think that should be fine to do, too?