Steffen-W/Import-LIB-KiCad-Plugin #51: Error while importing symbol.

jgardona, 2025-09-02:
Hello. Every time when I try to import a party from lcsc.com I get this error

```
Starting import for EasyEDA/LCSC component: C381116
Failed to import symbol: can't multiply sequence by non-int of type 'float'
Footprint already exists: DC-IN-TH_DC-005-5A-2.0.kicad_mod
3D model files already exist.
EasyEDA import completed, but no new files were created
```

Seems an issue in python script.

Version - 2025.08.25	169 kB	787 kB	✔	stable
Kicad Version - 9.4

[image]

[image]

[image]

It will not create de *_sym file to import in shchematics window.

Steffen-W, 2025-09-03:
https://github.com/Steffen-W/Import-LIB-KiCad-Plugin?tab=readme-ov-file#including-the-imported-libraries-in-kicad

Have you already read and followed the instructions in the readme file?

jgardona, 2025-09-03:
I only added this variable
KICAD_3RD_PARTY

The others not, because I was following another video in youtube also. I will try put the rest and test again.

closed 2025-09-05 by Steffen-W

sahlin, 2025-10-27:
I have the same problem sometimes. 

Starting import for EasyEDA/LCSC component: C167219
Failed to import symbol: can't multiply sequence by non-int of type 'float'
Footprint already exists: IND-SMD_L7.0-W6.6.kicad_mod
3D model files already exist.
EasyEDA import completed, but no new files were created

C167219 does not work. Others do.